espressif/esp-idf #785: MDC and MDIO pins undefined for Ethernet example in esp32

Nakul93, 2017-07-09:
CONFIG_PHY_SMI_MDC_PIN and CONFIG_PHY_SMI_MDIO_PIN declarations not found in Ethernet example.
Please let me know where should it be found?

krzychb, 2017-07-09:
Prefix `CONFIG_` usually points to the `menuconfig`. 

Looking inside `Kconfig.projbuild` there are:

- `PHY_SMI_MDC_PIN` - https://github.com/espressif/esp-idf/blob/3bd3640d60a0aff719470dd870d8ef3527ee9953/examples/ethernet/ethernet/main/Kconfig.projbuild#L42
- `PHY_SMI_MDIO_PIN` - https://github.com/espressif/esp-idf/blob/3bd3640d60a0aff719470dd870d8ef3527ee9953/examples/ethernet/ethernet/main/Kconfig.projbuild#L48

After running `make menuconfig`, an include file `sdkconfig.h` is being generated. 
Search your `build\include` directory for this file. 
Inside you should find declarations `CONFIG_PHY_SMI_MDC_PIN` and `CONFIG_PHY_SMI_MDIO_PIN`, you are looking for.

FayeY, 2017-07-17:
Hi @Nakul93, did the above suggestion help you resolve the issue?

Nakul93, 2017-07-17:
Yes it did thank you!!

closed 2017-07-17 by Nakul93