espressif/esp-idf #787: Cannot run SD example on ESP32

donny681, 2017-07-10:
> Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4400
load:0x40078000,len:11072
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

I (45) boot: ESP-IDF v3.0-dev-83-g14b0a360-dirty 2nd stage bootloader
I (45) boot: compile time 14:55:47
I (49) boot: Enabling RNG early entropy source...
I (65) boot: SPI Speed      : 40MHz
I (78) boot: SPI Mode       : DIO
I (90) boot: SPI Flash Size : 4MB
I (103) boot: Partition Table:
I (114) boot: ## Label            Usage          Type ST Offset   Length
I (137) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (160) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (183) boot:  2 factory          factory app      00 00 00010000 00100000
I (207) boot: End of partition table
I (220) boot: Disabling RNG early entropy source...
I (237) boot: Loading app partition at offset 00010000
I (563) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (564) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x04e04 ( 19972) map
I (580) boot: segment 2: paddr=0x00024e14 vaddr=0x3ffb0000 size=0x01b98 (  7064) load
I (609) boot: segment 3: paddr=0x000269b4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (633) boot: segment 4: paddr=0x00026dbc vaddr=0x40080400 size=0x07100 ( 28928) load
I (672) boot: segment 5: paddr=0x0002dec4 vaddr=0x400c0000 size=0x00000 (     0) load
I (685) boot: segment 6: paddr=0x0002decc vaddr=0x00000000 size=0x0213c (  8508) 
I (710) boot: segment 7: paddr=0x00030010 vaddr=0x400d0018 size=0x15c40 ( 89152) map
0x400d0018: _flash_cache_start at ??:?

I (736) cpu_start: Pro cpu up.
I (747) cpu_start: Starting app cpu, entry point is 0x40080cdc
0x40080cdc: call_start_cpu1 at /Users/sky/esp/esp-idf/components/esp32/./cpu_start.c:186

I (768) cpu_start: App cpu up.
I (780) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (802) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (823) heap_alloc_caps: At 3FFB23F0 len 0002DC10 (183 KiB): DRAM
I (843) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (864) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (886) heap_alloc_caps: At 40087500 len 00018B00 (98 KiB): IRAM
I (906) cpu_start: Pro cpu start user code
I (964) cpu_start: Starting scheduler on PRO CPU.
I (966) cpu_start: Starting scheduler on APP CPU.
I (966) example: Initializing SD card
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400e03a5  PS      : 0x00060d30  A0      : 0x80013bff  A1      : 0x3ffb3230  
0x400e03a5: configure_pin at /Users/sky/esp/esp-idf/components/driver/./sdmmc_host.c:307
 (inlined by) sdmmc_host_init_slot at /Users/sky/esp/esp-idf/components/driver/./sdmmc_host.c:335

A2      : 0x00000001  A3      : 0x00049060  A4      : 0xffffffff  A5      : 0xffffffff  
A6      : 0xffffffff  A7      : 0x00060023  A8      : 0x3ff49060  A9      : 0x00000004  
A10     : 0xffffffff  A11     : 0x8245f87c  A12     : 0x00000700  A13     : 0x3ffb40fc  
A14     : 0x3ffb2354  A15     : 0x00000002  SAR     : 0x00000036  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400e03a5:0x3ffb3230 0x40013bfc:0x3ffb3250 0x400df243:0x3ffb3290 0x400d08ba:0x3ffb3370
0x400e03a5: configure_pin at /Users/sky/esp/esp-idf/components/driver/./sdmmc_host.c:307
 (inlined by) sdmmc_host_init_slot at /Users/sky/esp/esp-idf/components/driver/./sdmmc_host.c:335

0x400df243: app_main at /Users/sky/esp/esp-idf/examples/storage/sd_card/main/./sd_card_example_main.c:48

0x400d08ba: main_task at /Users/sky/esp/esp-idf/components/esp32/./cpu_start.c:334

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4400
load:0x40078000,len:11072
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

The tip shows that the code sdmmc_host_init_slot() is error.

igrr, 2017-07-10:
Have you, by any chance, modified `host.slot` in the example to `SDMMC_HOST_SLOT_0`?
If not, are there any other modifications to the example code? What hardware are you using and how is the SD card connected to the ESP32?

FayeY, 2017-07-24:
Hi @donny681 , any update about this issue?

closed 2017-08-01 by FayeY

oaleeapp, 2017-12-12:
@igrr , I also faced this issue, and I want to ask what is the proper way to use `SDMMC_HOST_SLOT_0`, is slot 0 works fine?

igrr, 2017-12-12:
Depends on the pins used for connecting SPI flash. If SPI flash is connected to the default set of pins (6-11), SDMMC slot 0 can not be used.

If you connect SPI flash to a different set of pins (which will not overlap with SDMMC slot 0), then slot 0 will work.

oaleeapp, 2017-12-12:
Thanks for the prompt reply, it's really helpful and gives me some direction.

However, when I trying to implement, I encounter some issues as below:

```
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40103a63  PS      : 0x00060e30  A0      : 0x80103b77  A1      : 0x3ffbf2d0
0x40103a63: sdcard_init at /Users/lee/esp/development/OyaLabs/IoT_WifiManager_babyId/main/./main.c:710

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffb65d4
A6      : 0x00060021  A7      : 0x00000000  A8      : 0xffff8fff  A9      : 0x00003000
A10     : 0x00003b00  A11     : 0x0000001a  A12     : 0x3ff49028  A13     : 0x00000800
A14     : 0x0000001b  A15     : 0x00003b1c  SAR     : 0x00000006  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40133488  LEND    : 0x401334a7  LCOUNT  : 0x00000000
0x40133488: __divdf3 at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:1454

0x401334a7: __divdf3 at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:1481

```

When I try to `PIN_FUNC_SELECT(PERIPHS_IO_MUX_SD_DATA2_U, FUNC_SD_DATA2_HS1_DATA2);`, it crashed, which even track to very deep. 
Not sure how to solve this kind of problem.

Sincerely,

Victor

oaleeapp, 2017-12-12:
OK, got something from [here](https://esp32.com/viewtopic.php?t=1457):

```
Re: Crash when configure GPIO9
Postby ESP_Angus » Thu Mar 16, 2017 10:40 am

In common configurations (including ESP-WROOM32), GPIO9 is connected to the external SPI flash of the ESP32. So it can't easily be used as a general purpose I/O pin.

GPIOs 6, 7, 8, 9, 10, 11 are used this way on ESP-WROOM32.

I thought this was mentioned in the ESP-WROOM32 datasheet, but it seems to be overlooked. I'll request that the documentation team update the document.
```

TobeChan, 2018-10-22:
I can't download the code into esp32 when I pullup GPIOs  15, 2, 4, 12, 13  and connect GPIO0 with GPIO2
Can you tell me how to fix the problem?