espressif/esp-idf #788: Is it possible to make ESP32 emulate an SD card?

leopck, 2017-07-10:
I'm not well-verse in SD protocols but this is just a question I was wondering if it's possible.

Since ESP32 has SDIO capabilities, is it possible to make it not only read SD cards but to emulate an SD card? It has variety of use cases such as SD card over WiFi or OTA over WiFi SD card.

igrr, 2017-07-10:
Unfortunately, SDIO slave peripheral in the ESP32 does not support many of the commands required for SD memory cards.

In SDIO specification, IO_RW_EXTENDED command is used for data transfer. SD memory cards operate using different data transfer commands (READ_BLOCK_SINGLE, READ_BLOCK_MULTIPLE, WRITE_BLOCK_SINGLE, WRITE_BLOCK_MULTIPLE), plus there are a few other auxiliary commands which SD memory cards need to support.

closed 2017-07-10 by igrr

danielkucera, 2017-09-24:
@igrr but aren't this commands implemented in software?

igrr, 2017-09-24:
SDIO slave interface of the ESP32 handles all commands in hardware. Software only gets an interrupts when the host sends data transfer commands. Probing/initialization commands do not reach the software.

fk1000iq, 2025-02-13:
那用ESP32来实现模拟SD卡，从硬件直接支持是不行的喽？