espressif/esp-idf #791: Set static IP, NM & GW

fesch, 2017-07-10:
Hi

I'm trying to put a static IP address on the Ethernet interface (LAN8720 on an ESP32-EVB board from Olimex), but it does not work they I thought it would. The PHY still gets it address via DHCP :-(

Any idea what could be wrong?

```
int EthernetClass::begin(IPAddress local_ip, IPAddress subnet, IPAddress gateway)
{
	tcpip_adapter_init();
	
	tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA);
	
	tcpip_adapter_ip_info_t info;
    info.ip.addr = static_cast<uint32_t>(local_ip);
    info.gw.addr = static_cast<uint32_t>(gateway);
    info.netmask.addr = static_cast<uint32_t>(subnet);
	
	tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA, &info);
	
	esp_event_loop_init(&WiFiGenericClass::_eventCallback, NULL);
	
	if(tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA, &info) == ESP_OK) {
        // continue
    } else {
        return 0;
    }
	
	eth_config.phy_addr = ETH_PHY_ADDR;
    eth_config.gpio_config = ethernet_config_gpio;
    eth_config.tcpip_input = tcpip_adapter_eth_input;
    //eth_config.phy_power_enable = ethernet_power_enable;
    esp_err_t err = esp_eth_init(&eth_config);
    if(!err){
        err = esp_eth_enable();
		return 1;
    }
	
	return 0;
}
```

negativekelvin, 2017-07-10:
https://github.com/espressif/esp-idf/pull/657

hwmaier, 2017-07-11:
As @negativekelvin mentioned, I had the same issue and fixed it in pull request https://github.com/espressif/esp-idf/pull/657. I am not sure why this important but simple bug fix has not been adopted so far. 

Apply the PR to your local repository and you should be able to use static IPs.

fesch, 2017-07-12:
Recompiled the TCP/IP module, but my codes kickes me out when applying the IP settings. My previous code was also wrong, as I will nee to use the ETH adapter.

Does anyone spot my error?

```
int EthernetClass::begin(IPAddress local_ip, IPAddress subnet, IPAddress gateway)
{
	// connect the callback function
	esp_event_loop_init(&WiFiGenericClass::_eventCallback, NULL);

	// init the TCP/IP adapter
	tcpip_adapter_init();
	// stop the DHCP service
	tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_ETH);
	// create a new TCP/IP info 
	tcpip_adapter_ip_info_t info;
	/* reverse ??
    info.ip.addr = static_cast<uint32_t>(__builtin_bswap32(local_ip));
    info.gw.addr = static_cast<uint32_t>(__builtin_bswap32(gateway));
    info.netmask.addr = static_cast<uint32_t>(__builtin_bswap32(subnet));
	/**/
    info.ip.addr = static_cast<uint32_t>((local_ip));
    info.gw.addr = static_cast<uint32_t>((gateway));
    info.netmask.addr = static_cast<uint32_t>((subnet));
	// apply the IP settings
	if(tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_ETH, &info) == ESP_OK) {
        // continue
    } else {
        return 0;
    }
	
	eth_config.phy_addr = ETH_PHY_ADDR;
    eth_config.gpio_config = ethernet_config_gpio;
    eth_config.tcpip_input = tcpip_adapter_eth_input;
    //eth_config.phy_power_enable = ethernet_power_enable;
    esp_err_t err = esp_eth_init(&eth_config);
    if(!err){
        err = esp_eth_enable();
		return 1;
    }
	
	return 0;
}
```

fesch, 2017-07-12:
DHCP has not been stopped by the given command, so that's were I need to look at!

fesch, 2017-07-12:
DHCP has not been stopped, that's why the IP settings are not being applied :(

fesch, 2017-07-12:
The make process seams no to take into account my patched file. Need to check that ... but for today: enough ;)

fesch, 2017-07-16:
Editing the "right" file helped a lot!

#657 works, so please integrate it ....

closed 2017-07-16 by fesch

SuperFra, 2017-07-30:
Hello @fesch, thank you so much for your contribution to porting the ethernet library. I tested it and it works, but ... only in DHCP. i found your FIX about (applied patch to libtcpip_adapter to allow ETH to be used as interface), that override the libtcpip_adapter.a. So now I can use the static ip address but if I want to use the wifi I receive the following error: "assertion "netif_init != NULL" failed: file "C:/Users/robert.fisch/Documents/Arduino/sdk/msys32/home/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c", line 151, function: tcpip_adapter_start
abort() was called at PC 0x40111d93 on core 1 Guru Meditation Error: Core  1 panic'ed (abort)".
I hope it is a my mistake, what do you think about? Can you help me to solve this problem?
TIA

fesch, 2017-07-30:
I can confirm this, but while trying to upgrade to the latest ESP-IDF version, I got a lot of errors while compiling any example right now ... but I will check in some days again. I noticed some changes in libtcpip_adapter and think they cause the my recent problems ...

I also noticed strange things: while the same code does not run on my Olimex ESP32-EVB, it runs well on the LOLIN32 ... :-?

SuperFra, 2017-07-30:
Thanks for your fast reply. We all are in your hands. Please keep us up to date.

fesch, 2017-07-30:
You make me feel uncomfortable ;-)
But I ended up in finding the problem: while compiling the new library, I forgot to include & enable WiFi support. So I've uploaded a new version to my repository where WiFi works again.

https://github.com/fesch/arduino-esp32

SuperFra, 2017-07-31:
GREAT! Thanks a lot, everythings works now.