espressif/esp-idf #793: Wifi Init Crashes the Code

llewellynnu, 2017-07-11:
Hi ,

So my firmware stack has Uart1 Uart2 ADC , I2C , wifi and BLE running 
but when i initialise Wifi it crashes . Here is the log that it prints 

I (2488) wifi: wifi firmware version: 72ddf26
I (2488) wifi: config NVS flash: enabled
I (2488) wifi: config nano formating: disabled
I (2498) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2508) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2528) wifi: Init dynamic tx buffer num: 32
I (2528) wifi: Init dynamic rx buffer num: 32
I (2528) wifi: wifi driver task: 3ffdfd8c, prio:23, stack:4096
I (2538) wifi: Init static rx buffer num: 10
I (2538) wifi: Init dynamic rx buffer num: 32
I (2538) wifi: Init rx ampdu len mblock:7
I (2548) wifi: Init lldesc rx ampdu entry mblock:4
I (2548) wifi: wifi power manager task: 0x3ffe7928 prio: 21 stack: 2560
I (2558) Main: Setting WiFi configuration SSID...
ESP_ERROR_CHECK failed: esp_err_t 0x5 at 0x400e1b70
0x400e1b70: initialise_wifi at /Users/llewellyn/Documents/workspace/TestApp/main/./main.c:106 (discriminator 1)

file: "/Users/llewellyn/Documents/workspace/TestApp/main/./main.c" line 106
func: initialise_wifi
expression: esp_wifi_set_config(WIFI_IF_STA, &wifi_config)

Backtrace: 0x40088d7c:0x3ffd3510 0x400890f9:0x3ffd3530 0x400e1b70:0x3ffd3550 0x400e1c7b:0x3ffd3610 0x400e1c83:0x3ffd3630 0x400d2d42:0x3ffd3660

Can you let me know what could be the issue ?

negativekelvin, 2017-07-11:
Are you doing something different from https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_request/main/http_request_example_main.c#L75

llewellynnu, 2017-07-11:
No Looks the same . 
Here is my wifi init code 

void initialise_wifi(void) {
	tcpip_adapter_init();
	wifi_event_group = xEventGroupCreate();
	ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

	wifi_config_t wifi_config = { .ap = { .ssid = "Hello", .ssid_len = 0,
			.password = "test", .channel = 2, .authmode = WIFI_AUTH_WPA_PSK,
			.ssid_hidden = 0, .max_connection = 4, .beacon_interval = 100 } };

	for (int i = 0; i < sizeof(wifi_config.sta.ssid); i++) {
		wifi_config.sta.ssid[i] = (uint8_t) nubot.Ssid[i];
	}

	for (int i = 0; i < sizeof(wifi_config.sta.password); i++) {
		wifi_config.sta.password[i] = (uint8_t) nubot.Password[i];
	}

	printf("SSID %s PASSWORD %s", wifi_config.sta.ssid,
			wifi_config.sta.password);

	ESP_LOGI(LTAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
	ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
	ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
	ESP_ERROR_CHECK(esp_wifi_start());
}

HubbyGitter, 2017-07-11:
I think [this is code 5](https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_err.h#L35). It would be "NOTFOUND".

Are you sure it's a good idea to set `ssid_len` to 0?

negativekelvin, 2017-07-11:
You have to use different wifi_config_t for ap and sta they overlap (union)

llewellynnu, 2017-07-11:
Thanx for the promt response 
So i have updated the code to this void initialise_wifi(void) {

	tcpip_adapter_init();
	wifi_event_group = xEventGroupCreate();
	ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

	wifi_config_t wifi_config_ap = { .ap = { .ssid = "Hello", .ssid_len = sizeof("Hello"),
			.password = "test", .channel = 2, .authmode = WIFI_AUTH_WPA_PSK,
			.ssid_hidden = 0, .max_connection = 4, .beacon_interval = 100 } };

	wifi_config_t wifi_config_sta;

	for (int i = 0; i < sizeof(wifi_config_sta.sta.ssid); i++) {
		wifi_config_sta.sta.ssid[i] = (uint8_t) nubot.Ssid[i];
	}

	for (int i = 0; i < sizeof(wifi_config_sta.sta.password); i++) {
		wifi_config_sta.sta.password[i] = (uint8_t) nubot.Password[i];
	}

	ESP_LOGI(LTAG, "Setting WiFi configuration SSID %s...", wifi_config_sta.sta.ssid);
	ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config_sta));
	ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config_ap));
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
	ESP_ERROR_CHECK(esp_wifi_start());
}

It still gives me the same error

llewellynnu, 2017-07-11:
I Did some more testing and its the esp_wifi_set_config which throws the error. 
I tried initialising it either STA or AP mode and it throws the same error

HubbyGitter, 2017-07-11:
Where's your `esp_wifi_set_mode()` call?

I think the chance that this is a problem with the current or recent IDF WiFi code is pretty low, so at this point, I'd suggest that you move the thread over to the [ESP32 support forum](ESP32 support forum).

In general, people who spend their unpaid time supporting others will appreciate it if helpful information is provided, depending on the context. In this case, "the error" and "the same error" are really very unspecific and before your next post, you may want to look up the meaning of the returned error code in `esp_err.h` yourself and draw any conclusions.

llewellynnu, 2017-07-11:
@HubbyGitter Thanx for your reply . You can see the esp_wifi_set_mode() call in the code in previous comment . By "Same Error" I mean the error which is mentioned in the First comment .

HubbyGitter, 2017-07-11:
You must set the mode before passing the config, just as in the example code. (That's why I didn't see the  call.) You can figure why this is necessary if you look at the fact that you're filling the same data structure for both modes without providing any discriminator there.

llewellynnu, 2017-07-11:
@HubbyGitter Thanx alot it works now .

liuzfesp, 2017-07-17:
Hi @llewellynnu, the definition of ESP_ERROR_CHECK(): 
`#ifdef NDEBUG
#define ESP_ERROR_CHECK(x) do {                                         \
        esp_err_t rc = (x);                                             \
        (void) sizeof(rc);                                              \
    } while(0);
#else
#define ESP_ERROR_CHECK(x) do {                                         \
        esp_err_t rc = (x);                                             \
        if (rc != ESP_OK) {                                             \
            _esp_error_check_failed(rc, __FILE__, __LINE__,             \
                                    __ASSERT_FUNC, #x);                 \
        }                                                               \
    } while(0);
#endif
`

the crash is because esp_wifi_set_config returns 0x5, it means the wifi mode is wrong, then ESP_ERROR_CHECK assert, that's why you see the crash.

FayeY, 2017-07-23:
Hi @llewellynnu , did the above suggestion help you resolve the issue?

closed 2017-08-01 by FayeY

er-shubhendu-ee, 2021-07-05:
@liuzfesp Can you please guide where to find the error code id definitions are placed for this particular module?