espressif/esp-idf #794: BLE gatt_client app is not working properly

patilsan90, 2017-07-11:
I am trying to connect gatt_client to android device (moto g4 android 7.0). ESP is getting connected,
even i am able to transfter data also but after some time it does gets disconnects automatically.
I am getting below error 

E (40272) BT: btm_sec_clr_temp_auth_service() - no dev CB
E (40272) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
E (40272) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
E (40282) BT: Device not found
E (40282) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 47:44:c8:16:60:60

heyinling, 2017-07-21:
GATTC multiple connection is supported.

1. reg 1 gattc app
2. use gatt_open to create first connection, cache conn_id and remote address in connect event
3. use gatt_open to create second connection, cache conn_id and remote address in connect event

then you can use different conn_id and remote address with gattc APIs for different connections. Need to pay attention that you need to cache data for each connection (for example, service/char uuids got from discovery)

Weijian-Espressif, 2017-07-21:
@patilsan90 , The attachment is a multi_connect demo, you must have three slave device named "ESP_GATTS_DEMO_1", "ESP_GATTS_DEMO_2" and "ESP_GATTS_DEMO_3". 
[gattc_multi_connect.c.zip](https://github.com/espressif/esp-idf/files/1165413/gattc_multi_connect.c.zip)
[gatts_demo.c.zip](https://github.com/espressif/esp-idf/files/1165415/gatts_demo.c.zip)

FayeY, 2017-08-01:
Hi @patilsan90 , did the above suggestion help you resolve the issue?

patilsan90, 2017-08-01:
Yea.. thanks a lot

closed 2017-08-01 by FayeY

kukushonok1987, 2017-09-21:
Hello, I want to create more than 3 connection, please help to know which parameter in menuconfig should I change to do it?

nsamala, 2018-05-24:
@patilsan90 @kukushonok1987 I created a PR here to dynamically create connections to multiple servers. Check it out here: https://github.com/nkolban/ESP32_BLE_Arduino/pull/13