espressif/esp-idf #795: How to change the wifi MTU on esp32

liulingliuf, 2017-07-11:
Hi,
Recently, I'm doing a research that requires wifi to transmit data which is larger than 6000 **in one package**.  I already changed some setting in menuconfig and code, but it didn't work.
the output error by UART is:
E (77669) wifi: ebuf_free: ebuf corrupted type=1 eb=0x3ffb9930 ds_head=0x3ffb9970 desc=0x3ffb9994 begin=0x3ffc4e6c start=0x3ffc4e76 len=5056 size=982
I (78109) udp_perf:: try to read scrambler register and the register is:5

E (78129) wifi: esf_buf.c esf_buf_recycle 253

someone know how to fix this or how to change the MTU?@igrr@projectgus
thanks
Ling

hchaudhary1, 2017-07-15:
even if you could go to 6000, the the MTU is max 1500 on all other wifi/ethernet devices. How would communicate to anything else?

liuzfesp, 2017-07-24:
Hi @liulingliuf, generally we should NOT set so big MTU, otherwise we may encounter the issue @hchaudhary1 pointed out. On the other hand, from WiFi implementation perspective, we should NOT set too big MTU, let consider a scenario, the WiFi channel is very busying and the data rate in the air is degrade to 1Mbits, then a 6000Bytes packet requires (6000*8)/(1*1000)=48ms to be sent out, it means the device will occupy the channel for 48ms. If the packets is retransmitted by WiFi driver for 5 times, then it requires 48*5=240ms ... ... in other words, the WiFi may not work as expected if the MTU is too big in some special scenario. Anyway, I'm going to check it not work when MTU is 6000.

liuzfesp, 2017-07-24:
Hi @liulingliuf just check the implementation, currently the maximum MTU esp32 WiFi driver supports is 4017Bytes, however, we strong recommend that NOT modify the default MTU, otherwise unexpected behavior may happen...

liulingliuf, 2017-07-24:
@liuzfesp thanks for your reply. I do this for wifi research. As your point, the package with big MTU will take a long time to be sent out and occupy the channel for a long time. I use the wifi in the special scenario that there only two wifi devices communicating with each other. I want to know what limits the MTU bigger than 4017, how to change this. BTY, what the errors meaning followed: 
E (77669) wifi: ebuf_free: ebuf corrupted type=1 eb=0x3ffb9930 ds_head=0x3ffb9970 desc=0x3ffb9994 begin=0x3ffc4e6c start=0x3ffc4e76 len=5056 size=982
E (78129) wifi: esf_buf.c esf_buf_recycle 253

liuzfesp, 2017-07-24:
@liulingliuf the error because we have restriction for the TX packet length, when the packet is too long, WiFi driver packet sanity check will fail and assert the system and shows above error information. The max length 4017 is also because of the restriction in WiFi driver. We can support to modify MTU to any value, but it need some effort to do it:).

liulingliuf, 2017-07-24:
@liuzfesp thanks for your efforts. Waiting for the new SDK to support the features.

closed 2017-10-11 by FayeY

Hengxing-Tang, 2022-11-29:
someone can give me answer？

Xiehanxin, 2022-12-01:
@Hengxing-Tang https://github.com/espressif/esp-idf/blob/master/components/esp_netif/lwip/netif/wlanif.c#L44