espressif/esp-idf #798: mbed tls Malloc fails when BT is enabled

llewellynnu, 2017-07-12:
Hi , 
I am working on a product which involves . UART1 , UART2 , I2C , Wifi , BTLE , AWSIoT , PPPoS .
Everything works perfectly if BT is disabled . 
If i enable BT stack malloc in mbedtls for AWSIoT fails. 

Regards

negativekelvin, 2017-07-13:
https://github.com/espressif/esp-idf/issues/346

Check menuconfig for other memory saving options

FayeY, 2017-07-17:
Hi @llewellynnu , did the above suggestion help you resolve the issue?

llewellynnu, 2017-07-17:
Hi
I couldn't get everything running at the same time . Cause I couldn't free
up enough memory.

FHFS, 2017-07-24:
How big is your CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN? Decreasing this can free up a lot of memory.

llewellynnu, 2017-07-24:
Hi ,
I have reduced it to 5000

closed 2017-09-20 by llewellynnu