espressif/esp-idf #802: StoreProhibited on startup

jaracil, 2017-07-14:
StoreProhibited on last master branch.
commit: 4ec2abbf23084ac060679e4136fa222a2d0ab0e8

I (1727) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1747) heap_init: At 400943D8 len 0000BC28 (47 KiB): IRAM
Guru Meditation Error of type StoreProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4000c2e4  PS      : 0x00060530  A0      : 0x800d3b9d  A1      : 0x3ffe3840  
A2      : 0x00000000  A3      : 0x3ffe3860  A4      : 0x000000e0  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x0000000c  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x400943d8  A11     : 0x0000bc04  A12     : 0x3ffe376f  A13     : 0x00000037  
A14     : 0x00000000  A15     : 0x3ffe3773  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x0000000d  

Backtrace: 0x4000c2e4:0x3ffe3840 0x400d3b9a:0x3ffe3850 0x40081249:0x3ffe3c60 0x400787ea:0x3ffe3ca0 0x400789f3:0x3ffe3cd0 0x40078f81:0x3ffe3d50 0x400800f6:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x400d3b9a: heap_caps_init at /home/pepe/esp32/esp-idf/components/heap/./heap_caps_init.c:173

0x40081249: call_start_cpu0 at /home/pepe/esp32/esp-idf/components/esp32/./cpu_start.c:170

0x400800f6: _WindowUnderflow8 at ??:?

Rebooting...

igrr, 2017-07-18:
What program do you get this issue with? Could you please post a link to your sdkconfig file?

projectgus, 2017-07-18:
In addition to @igrr's request, could you please paste the full boot log from reset? I think the problem is to do with the heap configuration (which is influenced by sdkconfig and by your app's static memory usage). The boot log includes a full dump of the heap layout (only the last few lines are visible in the bug report above).

jaracil, 2017-07-19:
@projectgus Here is complete dump:

I (48) boot: ESP-IDF v1.0-1757-gd998bfc7-dirty 2nd stage bootloader
I (48) boot: compile time 08:41:38
I (48) boot: Enabling RNG early entropy source...
I (68) boot: SPI Speed      : 40MHz
I (81) boot: SPI Mode       : DIO
I (93) boot: SPI Flash Size : 4MB
I (105) boot: Partition Table:
I (117) boot: ## Label            Usage          Type ST Offset   Length
I (139) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (163) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (186) boot:  2 factory          factory app      00 00 00010000 00100000
I (209) boot:  3 access           unknown          40 00 00110000 00080000
I (232) boot: End of partition table
I (245) boot: Disabling RNG early entropy source...
I (262) boot: Loading app partition at offset 00010000
I (1391) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (1391) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x29920 (170272) map
I (1408) boot: segment 2: paddr=0x00049930 vaddr=0x3ffc0000 size=0x02714 ( 10004) load
I (1439) boot: segment 3: paddr=0x0004c04c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (1462) boot: segment 4: paddr=0x0004c454 vaddr=0x40080400 size=0x1405c ( 82012) load
I (1527) boot: segment 5: paddr=0x000604b8 vaddr=0x400c0000 size=0x00000 (     0) load
I (1528) boot: segment 6: paddr=0x000604c0 vaddr=0x00000000 size=0x0fb48 ( 64328) 
I (1546) boot: segment 7: paddr=0x00070010 vaddr=0x400d0018 size=0x6210c (401676) map
0x400d0018: _stext at ??:?

I (1573) cpu_start: Pro cpu up.
I (1584) cpu_start: Starting app cpu, entry point is 0x400810a4
0x400810a4: call_start_cpu1 at /home/pepe/esp32/esp-idf/components/esp32/./cpu_start.c:192

I (1605) cpu_start: App cpu up.
I (1618) heap_init: Initializing. RAM available for dynamic allocation:
I (1638) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1657) heap_init: At 3FFB3000 len 00005000 (20 KiB): DRAM
I (1676) heap_init: At 3FFBBB28 len 00002000 (8 KiB): DRAM
I (1695) heap_init: At 3FFCF708 len 000108F8 (66 KiB): DRAM
I (1715) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1734) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1754) heap_init: At 4009445C len 0000BBA4 (46 KiB): IRAM
Guru Meditation Error of type StoreProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4000c2e4  PS      : 0x00060530  A0      : 0x800d3bed  A1      : 0x3ffe3860  
A2      : 0x00000000  A3      : 0x3ffe3880  A4      : 0x000000e0  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x0000000c  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x4009445c  A11     : 0x0000bb80  A12     : 0x3ffe378f  A13     : 0x00000036  
A14     : 0x00000000  A15     : 0x3ffe3793  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x0000000d  

Backtrace: 0x4000c2e4:0x3ffe3860 0x400d3bea:0x3ffe3870 0x4008121d:0x3ffe3c80 0x400787f2:0x3ffe3ca0 0x40078a23:0x3ffe3cd0 0x40078fad:0x3ffe3d50 0x400800f6:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x400d3bea: heap_caps_init at /home/pepe/esp32/esp-idf/components/heap/./heap_caps_init.c:173

0x4008121d: call_start_cpu0 at /home/pepe/esp32/esp-idf/components/esp32/./cpu_start.c:177

0x400800f6: _WindowUnderflow8 at ??:?

Rebooting...

jaracil, 2017-07-19:
@igrr here is my sdkconfig:

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python2"

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL=3

#
# Security features
#

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_DIO=y
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=y
CONFIG_ESPTOOLPY_AFTER="hard_reset"
CONFIG_MONITOR_BAUD_115200B=y
CONFIG_MONITOR_BAUD_OTHER_VAL=115200
CONFIG_MONITOR_BAUD=115200

#
# Partition Table
#
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_CUSTOM_APP_BIN_OFFSET=0x10000
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
CONFIG_APP_OFFSET=0x10000

#
# Compiler options
#
CONFIG_OPTIMIZATION_LEVEL_DEBUG=y
CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED=y

#
# Component config
#

#
# Application Level Tracing
#
CONFIG_ESP32_APPTRACE_DEST_NONE=y

#
# FreeRTOS SystemView Tracing
#
CONFIG_BT_ENABLED=y
CONFIG_BLUEDROID_ENABLED=y
CONFIG_BTC_TASK_STACK_SIZE=3072
CONFIG_BT_DRAM_RELEASE=y
CONFIG_GATTS_ENABLE=y
CONFIG_BLE_SMP_ENABLE=y
CONFIG_BT_ACL_CONNECTIONS=7
CONFIG_BTDM_CONTROLLER_RUN_CPU=0
CONFIG_SMP_ENABLE=y
CONFIG_BT_RESERVE_DRAM=0x10000

#
# ESP32-specific
#
CONFIG_ESP32_DEFAULT_CPU_FREQ_160=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=160
CONFIG_MEMMAP_SMP=y
CONFIG_TRACEMEM_RESERVE_DRAM=0x0
CONFIG_ESP32_ENABLE_COREDUMP_TO_NONE=y
CONFIG_FOUR_UNIVERSAL_MAC_ADDRESS=y
CONFIG_NUMBER_OF_UNIVERSAL_MAC_ADDRESS=4
CONFIG_SYSTEM_EVENT_QUEUE_SIZE=32
CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE=4096
CONFIG_MAIN_TASK_STACK_SIZE=4096
CONFIG_IPC_TASK_STACK_SIZE=1024
CONFIG_NEWLIB_STDOUT_ADDCR=y
CONFIG_CONSOLE_UART_DEFAULT=y
CONFIG_CONSOLE_UART_NUM=0
CONFIG_CONSOLE_UART_BAUDRATE=115200
CONFIG_ULP_COPROC_RESERVE_MEM=0
CONFIG_ESP32_PANIC_PRINT_REBOOT=y
CONFIG_ESP32_DEBUG_OCDAWARE=y
CONFIG_INT_WDT=y
CONFIG_INT_WDT_TIMEOUT_MS=300
CONFIG_INT_WDT_CHECK_CPU1=y
CONFIG_TASK_WDT=y
CONFIG_TASK_WDT_TIMEOUT_S=5
CONFIG_TASK_WDT_CHECK_IDLE_TASK=y
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y
CONFIG_BROWNOUT_DET=y
CONFIG_BROWNOUT_DET_LVL_SEL_0=y
CONFIG_BROWNOUT_DET_LVL=0
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
CONFIG_ESP32_RTC_CLOCK_SOURCE_INTERNAL_RC=y
CONFIG_ESP32_RTC_CLK_CAL_CYCLES=1024
CONFIG_ESP32_DEEP_SLEEP_WAKEUP_DELAY=0
CONFIG_ESP32_XTAL_FREQ_AUTO=y
CONFIG_ESP32_XTAL_FREQ=0
CONFIG_PHY_ENABLED=y

#
# PHY
#
CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE=y
CONFIG_ESP32_PHY_MAX_TX_POWER=20

#
# FAT Filesystem support
#
CONFIG_FATFS_CODEPAGE_ASCII=y
CONFIG_FATFS_CODEPAGE=1
CONFIG_FATFS_MAX_LFN=255

#
# FreeRTOS
#
CONFIG_FREERTOS_CORETIMER_0=y
CONFIG_FREERTOS_HZ=100
CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION=y
CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY=y
CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=1
CONFIG_FREERTOS_ASSERT_FAIL_PRINT_CONTINUE=y
CONFIG_FREERTOS_BREAK_ON_SCHEDULER_START_JTAG=y
CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=1024
CONFIG_FREERTOS_ISR_STACKSIZE=1536
CONFIG_FREERTOS_MAX_TASK_NAME_LEN=16
CONFIG_TIMER_TASK_PRIORITY=1
CONFIG_TIMER_TASK_STACK_DEPTH=2048
CONFIG_TIMER_QUEUE_LENGTH=10

#
# Log output
#
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL=3
CONFIG_LOG_COLORS=y

#
# LWIP
#
CONFIG_LWIP_MAX_SOCKETS=10
CONFIG_LWIP_THREAD_LOCAL_STORAGE_INDEX=0
CONFIG_LWIP_DHCP_MAX_NTP_SERVERS=1

#
# TCP
#
CONFIG_TCP_MAXRTX=12
CONFIG_TCP_SYNMAXRTX=6
CONFIG_TCP_MSS=1436
CONFIG_TCP_SND_BUF_DEFAULT=5744
CONFIG_TCP_WND_DEFAULT=5744
CONFIG_TCP_QUEUE_OOSEQ=y
CONFIG_TCP_OVERSIZE_MSS=y
CONFIG_LWIP_DHCP_DOES_ARP_CHECK=y
CONFIG_TCPIP_TASK_STACK_SIZE=2048

#
# ICMP
#

#
# Virkey.com config
#
CONFIG_VK_BOARD_CUSTOM is not set
CONFIG_VK_BOARD_OLIMEX_EVB=y

#
# mbedTLS
#
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=16384
CONFIG_MBEDTLS_HAVE_TIME=y

#
# OpenSSL
#
CONFIG_OPENSSL_ASSERT_DO_NOTHING=y

#
# SPI Flash driver
#
CONFIG_SPI_FLASH_ROM_DRIVER_PATCH=y

projectgus, 2017-07-19:
Thanks @jaracil . I can see the bug (the heap allocator stores some of its metadata in the first region when setting up, and the first heap is only 0xF0 bytes in this case so it's too small to hold it all.)

A fix should be available soon.

danicampora, 2017-07-19:
@projectgus I'm hitting this issue as well :-(. Is there an ETA for the fix?

closed 2017-07-21 by igrr

perll, 2018-03-30:
Hi, all!
Some news about that fix?
Regards

projectgus, 2018-04-03:
hi @perll,

This should be fixed in the commit 0feb408 linked just above your comment. This fix is part of v3.0-rc1 & the master v3.1 development branch (you can click on the commit to see which releases it is included in.)

Are you still seeing this issue despite the fix?

Angus

perll, 2018-04-12:
@projectgus , Sir, sorry for the delay.
Yes, I'm on last master branch but, the issues still happening.
Now, after get the master Updated a day ago by igrr , the GATT_CLIENT, joinned with HTTP_REQUEST do not work for more tha 10'
Regards

projectgus, 2018-04-12:
>  the GATT_CLIENT, joinned with HTTP_REQUEST do not work for more tha 10'

This sounds like it might be something different to the issue shown above (crashing instantly on startup). Could you please open a new issue and attach some serial output logs, and link to source code if possible? Thanks.

perll, 2018-04-12:
Sure!
I'm on contact with @TimXia . I've been sent ELF and SDKCONFIG and logs... this stuffs. For now, I'm waiting for next git deploy.
TKU!

perll, 2018-04-12:
And now... same ELF, same SDKCONFIG with new master build:

I (105) boot: End of partition table
E (109) boot: ota data partition invalid, falling back to factory
I (116) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x36bd0 (224208) map
I (203) esp_image: segment 1: paddr=0x00046bf8 vaddr=0x3ffc0000 size=0x03ae8 ( 15080) load
I (209) esp_image: segment 2: paddr=0x0004a6e8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/raimundo/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (210) esp_image: segment 3: paddr=0x0004aaf0 vaddr=0x40080400 size=0x05520 ( 21792) load
I (228) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xb2fd0 (733136) map
0x400d0018: _stext at ??:?

I (485) esp_image: segment 5: paddr=0x00102ff0 vaddr=0x40085920 size=0x0e314 ( 58132) load
0x40085920: r_lld_evt_restart at ??:?

I (509) esp_image: segment 6: paddr=0x0011130c vaddr=0x400c0000 size=0x00000 (     0) load
E (510) esp_image: Image length 1053488 doesn't fit in partition length 1048576
E (515) boot: Factory app partition is not bootable
E (521) esp_image: image at 0x110000 has invalid magic byte
E (527) boot: OTA app partition slot 0 is not bootable
E (533) esp_image: image at 0x210000 has invalid magic byte
W (539) esp_image: image at 0x210000 has invalid SPI mode 255
W (545) esp_image: image at 0x210000 has invalid SPI size 15
E (552) boot: OTA app partition slot 1 is not bootable
E (557) boot: No bootable app partitions in the partition table
user code done

projectgus, 2018-04-12:
> E (510) esp_image: Image length 1053488 doesn't fit in partition length 1048576

The app is too big for the default partition table (1MB partitions). If you can't make it smaller, you can set a custom partition table with a larger partition:
http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html

perll, 2018-04-12:
I've been use this code on las 3 months. I cant understand what was changed
after git master  update.
My app is GATT_CLIENTE and HTTP_CLIENT joined.
Regards!

projectgus, 2018-04-12:
If you updated IDF then something may have added a small amount of code size which pushed the app over 1MB in size.

If you want to continue this discussion, the esp32.com forum is a good place to post. If you can share the code, it may be possible to find ways to reduce the code size and figure out the cause of the crash.

perll, 2018-04-12:
according to what I was told, Bluetooth and WiFi do not work well together.
after several changes, I still have problems and my project is not robust.
I need to rethink the physical and logical architecture. Thank you very
much for your attention.