espressif/esp-idf #805: Feature request: UART code improvement

samsam4, 2017-07-15:
In case of UART_PATTERN_DET event, the uart_event_t.size  to return the position in the buffer of the pattern detected(i.e. size of the buffered data before and incl. the terminators) instead of the total data size in the buffer.
It is a common situation to transmit via uart some chunks of data or commands which are terminated with a predefined terminator char(or few chars). Before the terminator is received, the part of data/command received and accumulated in the uart buffer so far  is useless and is waste of resources to grab and process every few chars when they come available instead on UART_PATTERN_DET event notification to pull only the finished(terminated) chunk and whatever bytes left in the buffer to wait there until next terminator come.
Else with current condition need one extra buffer to move the data from the uart there, locate the terminator, use the bytes before it, then delete them and shift if there some leftovers to wait for the next chunk(s) of data until next terminator is available and so on .. :(

FayeY, 2017-11-28:
Hi @samsam4 , this feature should have been added in the latest IDF. If it is not, please feel free to reopen this issue. Thanks.

projectgus, 2018-01-12:
Reopening as this feature is not part of ESP-IDF yet.

closed 2018-01-29 by FayeY