espressif/esp-idf #806: Feature request: Ring Buffer code improvement

samsam4, 2017-07-15:
In case of ring buffer created in RINGBUF_TYPE_ALLOWSPLIT mode, currently there is no way to detect if the item saved in the ring buffer  was split in 2 chunks (i.e. when the place at the end of the buffer was not enough to place there the whole chunk, so the data stream have to roll over to the empty/available  space at the beginning of the buffer). Also in this mode there is no way to find the free space available in the buffer as the xRingbufferGetMaxItemSize returns same fixed value based on the size of the buffer when was created.
Proposed solution: every record/item put in the ring buffer in RINGBUF_TYPE_ALLOWSPLIT mode already have an extra head add which is size_t (i.e. unsigned long int) where currently stay only the size/length of the item. Just use MSB there as flag if there is a split. 
Now - to obtain this flag - few ways, but probably easiest with minimum changes would be 
void *xRingbufferReceive(RingbufHandle_t ringbuf, **long** *item_size, TickType_t ticks_to_wait);
where I changed the  item_size from unsigned long to long, so when get negative result there, will know that have to proceed with one extra  xRingbufferReceive in order to get the rest part of the original chunk?
And will be nice xRingbufferGetMaxItemSize to return the free space in the buffer when in RINGBUF_TYPE_ALLOWSPLIT mode or to have some extra function to do so?

samsam4, 2017-07-17:
Oops - err in last line of my previous post: 
...And will be nice xRingbufferGetMaxItemSize to return the free space in the buffer when in RINGBUF_TYPE_**NOSPLIT** mode or to have some extra function to do so?

The problem is that in RINGBUF_TYPE_NOSPLIT mode, xRingbufferGetMaxItemSize  returns a fixed size, which is  (buffer_size/2)-8 bytes, but far more important at runtime is what is the free space in the buffer ..

FayeY, 2017-12-26:
Hi, this issue should have been solved. If it is not, please feel free to reopen it. Thanks.

closed 2017-12-26 by FayeY