espressif/esp-idf #808: user/custom space adderss on the SPI flash for storing data

msharathreddy, 2017-07-15:
hi,
A single esp32's flash contain multiple apps,as well as many different kinds of data(calibration data,file systems,parameter storage and etc).
I'll try to know these addresses,allocated and unallocated memory(free space) locations in the SPI flash so that we can maintain partition mechanism.But I did not found yet. 
For this what can i do to resolve the above problem.
can you people help us on this issue.sorry my English.

regards
sharath

hchaudhary1, 2017-07-15:
look in:
`/esp-idf/components/partition_table`
there are the csv files there for the partition info

krzychb, 2017-07-17:
Optionally refer to documentation ;) 

> I'll try to know these addresses,allocated and unallocated memory(free space) locations in the SPI flash

For standard partitioning the addresses are in http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html#built-in-partition-tables

Unallocated space is then the function of SPI flash size of your ESP32 board.

closed 2017-08-09 by FayeY