espressif/esp-idf #809: driver/rmt: Bidirectional use case

devsaurus, 2017-07-16:
One of my use cases for rmt is to have an RX and a TX channel connected both to one single pin for implementing bidirectional protocols like Dallas' 1-wire. The TX channel initiates a transaction while RX channel stands by, triggers on activity and samples the incoming bit stream.

This worked reasonably well until [gpio driver: Clear GPIO matrix output config when setting pin as input](https://github.com/espressif/esp-idf/commit/17747b33586075e46cadfe0c096dabb7f1a80df2) was applied. The situation is now as follows:
- `rmt_set_pin()` calls `gpio_set_direction()` either for input or output
- `gpio_set_direction()` enables the desired direction but strictly disables the opposite path.
- A call to `rmt_set_pin()` for the RX channel undoes a previous `rmt_set_pin()` for the TX channel and vice versa. There's no way to have both directions enabled at the same time.

From my point of view the GPIO matrix and input enable handling within `rmt_set_pin()` should be less intrusive: Just route input path to RX channel or route TX channel to output drivers. But do not interfere with the opposite direction.

negativekelvin, 2017-07-16:
So add RMT_MODE_RXTX? Or do you also want the ability to set an rx and tx from different channels to the same pin?

devsaurus, 2017-07-17:
The latter: Connect an rx and a tx channel to the same pin.

I'm not aware that a single rmt channel could handle both at the same time.

negativekelvin, 2017-07-17:
> The RMT module contains eight channels; each channel has a transmitter and receiver, of which one can be active per channel.

Good point

suda-morris, 2022-05-16:
biderection configuration is now supported by the new RMT driver if you enable the [io_loop_back](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#_CPPv4N23rmt_tx_channel_config_t12io_loop_backE) flag of `rmt_tx_channel_config_t` and `rmt_rx_channel_config_t`. This feature is also widely used internally in esp-idf test code (to simplify the test runner setup). You can find the example [here](https://github.com/espressif/esp-idf/blob/master/tools/unit-test-app/components/test_utils/ref_clock_impl_rmt_pcnt.c#L85) and [here](https://github.com/espressif/esp-idf/blob/master/components/driver/test_apps/rmt/main/test_rmt_rx.c#L51)

closed 2022-05-16 by suda-morris