espressif/esp-idf #811: Simple Bluetooth Example

alexgrauer, 2017-07-17:
Hi all,

I open this thread because im having problems to make BLE work.
Im using the gatt_server_service_table example, but im not being able to change the read value of Heart Rate Control Point.

Im doing the following:

In the gatts_profile_event_handler i added:
case ESP_GATTS_READ_EVT:
heart_ctrl_point[1] += 0x01;
ESP_LOGI(GATTS_TABLE_TAG, "READ EVENT: %d\n",heart_ctrl_point[1]);
break;

When i read the value using the nRF Connect App i can see the READ EVENT text in the monitor, and the heart_ctrl_point is actually being incremented.
But in the App i still see 0x00 every time i read it.

Any clue?

Thanks in advance,
Alex

mh-dev, 2017-07-17:
This depends on if you're using auto response or not. There is an api to set attribute values which is what you should use it is `esp_ble_gatts_set_attr_value`.
Setting it in the read event does probably not what you want. Since I expect the value is already defined at this point. Another thing is that `esp_ble_gatts_set_attr_value` is  an async operation.

FayeY, 2017-07-20:
Hi @alexgrauer , did the above suggestion help you resolve the issue?

alexgrauer, 2017-07-20:
Hi @FayeY, I wasnt able to solve it yet. Im not sure if im calling the esp_ble_gatts_set_attr_value correctly. Im not sure i understand the parameters that go with it.

@mh-dev, if i set it in the read event, should i read the new value the next time i read it?

mh-dev, 2017-07-20:
You should checkout https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server/main/gatts_demo.c#L271-L281

Weijian-Espressif, 2017-07-21:
you shoud do like this :
 
case ESP_GATTS_READ_EVT:
            heart_ctrl_point[0] += 0x01;
            esp_ble_gatts_set_attr_value(param->read.handle, sizeof(heart_ctrl_point), heart_ctrl_point);
            ESP_LOGI(GATTS_TABLE_TAG, "READ EVENT: %d\n",heart_ctrl_point[0]);
       	 break;
The attachment is the test demo
[gatts_table_creat_demo.c.zip](https://github.com/espressif/esp-idf/files/1165373/gatts_table_creat_demo.c.zip)

alexgrauer, 2017-07-22:
Thanks very much @Weijian-Espressif, that worked!! 
Now, if i want to change heart_ctrl_point´s value from another function, not from the READ_EVT, what should i put instead of param->read.handle.

And also, is it possible to make heart_ctrl_point uint32 instead of uint8?

Thanks again

Weijian-Espressif, 2017-07-24:
@alexgrauer  maybe you can refer to gatts_demo.c , you can use  ble app test, you can write to some data to the demo, and after  your app enable demo's notify, the demo would send you some data. 
[gatts_demo.c.zip](https://github.com/espressif/esp-idf/files/1168725/gatts_demo.c.zip)
 making heart_ctrl_point uint32 is ok.

closed 2017-08-07 by FayeY