espressif/esp-idf #814: ESP32 Low Noise Amplifier

lmagalhaes1337, 2017-07-18:
Hi there, is there a way to use this amplifier and change the gain ?

igrr, 2017-07-18:
There is no driver for LNA yet; some work has been done to support this, but not ready for release.

The gain can be changed in software only by changing ADC attenuation. The gain of LNA itself is set by the external sampling capacitors connected to sens_vp / sens_vn pins, and can not be changed in software.

hberjanin, 2017-12-26:
hello,

do you have a stimated time to arrival for the info/examples of the LNA?

Thanks!!

akofoed, 2018-01-08:
I would like an example. I have tried different things as described in the documentation but I cannot make it amplify the input signal.
Any examples?

javicordon, 2018-01-26:
Hi, are there any updates on support for LNA?
Thanks!

hydrofields, 2018-05-26:
yes no support for LNA

areguy, 2019-04-10:
is there API to get the gain?

Nufflee, 2019-07-16:
Any updates on this @igrr? It's been two years already

JosuGZ, 2021-02-02:
> Any updates on this @igrr? It's been two years already

It seems like it was removed from the technical reference:

[image]

You can check this link: https://github.com/krzychb/esp32-lna

ShrinivasNirety, 2022-01-21:
Any update on this?