espressif/esp-idf #815: "emac rx buf err!! in esp_eth_free_rx_buf

alarruskain, 2017-07-18:
Hi, 
I'm facing a deadlock "emac rx buf err!!" loop, called from esp_eth_free_rx_buf,, since I've synced to commit d724cc23d27177376c265a728f86a41619396e51
Might there be something incorrect?
regards

liuzfesp, 2017-07-20:
HI @alarruskain currently the fix is in review status, feel free to let me know if you find it doesn't work: [https://github.com/liuzfesp/esp-idf/commit/37033761c59b40932b21801a1c338eeab18c6c59](url)

alarruskain, 2017-07-21:
I've tested your release and there are not buffer errors any more. 
Thank you

closed 2017-07-25 by alarruskain

comcomservices, 2018-06-21:
Hi @alarruskain @liuzfesp 

We're experiencing the same issue since moving to the latest master branch from the 2.xx release

D (3624507) phy_common: phy_mii_check_link_status(UP)
D (3626508) phy_common: phy_mii_check_link_status(UP)
D (3628507) phy_common: phy_mii_check_link_status(UP)
D (3630507) phy_common: phy_mii_check_link_status(UP)
D (3631320) emac: tx buf full

MAC communication is working, all registers read correctly etc however it seems that the interface does not ever get an IP or provide any communication. the link status changes correctly as a cable is connected and disconnected but not packets are exchanged anymore since the update..

We are using a DP83848 PHY and wrote a custom handler for it