espressif/esp-idf #816: Can't compile OTA sample in cpp

thecashit, 2017-07-18:
Hello!

Tried to compile OTA in CPP project and has linker error. 
libapp_update.a(esp_ota_ops.o):(.literal.esp_ota_end+0x0): undefined reference to `esp_image_basic_verify'
libapp_update.a(esp_ota_ops.o): In function `esp_ota_set_boot_partition':
components/app_update/esp_ota_ops.c:359: undefined reference to `esp_image_basic_verify'

How can i solve this?

Thank you!

HubbyGitter, 2017-07-18:
Always the same procedure. Check whether the .h file where the function is declared contains the `extern "C"` magic and if not, figure which `#include` in your source code can be surrounded by `extern "C" { }` to compensate it.

thecashit, 2017-07-18:
Thnx! Hope it helps.
Try to do this short time.

thecashit, 2017-07-19:
Hm...
I tried a lot of options but I always encounter an error.

>> Always the same procedure.
Correct.

But what shall i do when i need to include only "esp_ota_ops.h" with esp_ota_end function() inside and have error "undefined reference to `esp_image_basic_verify'". esp_image_basic_verify is in "esp_image_format.h" that included inside esp_ota_ops.h.

extern "C" { #include ""esp_ota_ops.h" } don't solve linker error.

Is exists any variants besides of edit components/bootloader_support/* files?

Thnx!

HubbyGitter, 2017-07-19:
While `esp_image_format.h` does not have the `extern "C` magic, it _does_ have protection against being included repeatedly.

So it's definitely worth trying

    extern "C" {
    #include "esp_image_format.h"
    }
    #include "esp_ota_ops.h"

But tbh I haven't analysed the include dependency graph nor have I checked why the linker actually fails. So if that also fails, you'll simply first have to analyse the root cause.

There's almost always a way; it's just a matter of understanding what is included where, and in some cases you need to include more than just the header in question before the `#include` which causes the problem.

thecashit, 2017-07-19:
I tried to do this already. But nothing changed.

Maybe there are some error in code, maybe not...
In any case, thanks for the help!

I will do a deeper study of the structure of the components. You gave me an understanding of where to move next

FHFS, 2017-07-19:
I run the ota example in cpp with minor modifications and no extern "C" guards around any of the includes. Can you post your cpp example?
My modifications are:
`extern "C" void app_main()`
and in initialise_wifi()
```
wifi_config_t wifi_config;
memset(&wifi_config, 0, sizeof(wifi_config));
strcpy((char *)wifi_config.sta.ssid, EXAMPLE_WIFI_SSID);
strcpy((char *)wifi_config.sta.password, EXAMPLE_WIFI_PASS);
```

thecashit, 2017-07-19:
@FHFS Thnx for answer.

That's not a typical cpp example.
I try to make OTA_t class to use it in my app.
I attached a zip file with the app's snippet.

There is no Wi-Fi initialization, however, at the linking stage it is not required, as far as I understand.
The code may not work, but to debug i need to start firmware on ESP32 and can't do this now.
Project under the Platformio IDE.

I would be grateful for any help

[OTA_t.zip](https://github.com/espressif/esp-idf/files/1158857/OTA_t.zip)

FHFS, 2017-07-19:
I've coppied your src folder to my project and it compiles and runs O_o
[log](https://pastebin.com/VZjgBySN)

I suggest you look into which esp-idf commit platformIO uses.

thecashit, 2017-07-19:
@FHFS
0_o ..i spend for hours to understand whats wrong.
I think you're right - as i see in your log you use "ESP-IDF v3.0-dev" framework version.
Platformio uses 2.2.

Thank you!
I'lI didn't notice that the third version was released.

Now it is clear what the problem is!

FayeY, 2017-07-20:
Is this problem solved now?

thecashit, 2017-07-20:
@FayeY 
Solved.
To fix that it need to remove "bootloader_support" from ignore_dirs variable in platformio espidf.py script.

closed 2017-07-20 by thecashit

ok-net, 2017-12-01:
Hi,  thecashit

I have this exact problem.  I'm new to the build environment with Platformio. Can you give more details on your fix?  Where is the change made (files\settings) made?  Thanks

thecashit, 2017-12-04:
@ok-net
It's been a long time...
Try to update your Platformio frameworks with "platformio update" command in terminal