espressif/esp-idf #817: SD Card example: A fatal error occurred: Timed out waiting for packet header

Staubgeborener, 2017-07-18:
I tried the SD Card example and get the following error messages:
```
$ make flash monitor
Flashing binaries to serial port COM3 (app at offset 0x10000)...
esptool.py v2.0-beta3
Connecting........_
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Flash params set to 0x0220
Compressed 15824 bytes to 9216...
Wrote 15824 bytes (9216 compressed) at 0x00001000 in 0.8 seconds (effective 154.7 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
make: *** [/home/Staubgeborener/esp/esp-idf/components/esptool_py/Makefile.projbuild:53: flash] Fehler 2
```

[This is my circuit](http://i.imgur.com/f9sHfFo.png), taken from [this](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card) page.

llewellynnu, 2017-07-18:
IO2 needs to be pulled down to go into programming mode. While Programming short IO2 and IO0.

llewellynnu, 2017-07-18:
That error can also occur because IO12 is pulled high. IO12 is used for selecting the voltage to the Flash memory read the note in the readme of the SD card example

Staubgeborener, 2017-07-18:
I will try a pulldown resistor for IO2. But what do you mean with "while programming short io2 and io0?"

llewellynnu, 2017-07-18:
While programming If u look at the Sch of the development board. RTS and DTR are used to pull the EN and IO0 down for programming. So if u short IO0 and IO2 then IO2 will also be pulled down along with IO0

llewellynnu, 2017-07-18:
But I still feel it's an issue with IO12. Disconnect IO12 and see if it works. If it does work then you will have to set the efuse to program the voltage to flash to be 3.3V

Staubgeborener, 2017-07-18:
First of all: Thanks for taking the time and helping me!
[I try to find the IO0, but can't find it.](http://i.imgur.com/tQ8jI4W.jpg)

I also tried to disconnect IO12 from the SD Card (and to connect IO12 to SD Card but without pull up resistor), but the error message will be the same.

Staubgeborener, 2017-07-18:
First of all: Thanks for taking the time and helping me @llewellynnu ! 

I bought the esp32 from doit.am and there is no IO0. I'm still looking with the pinout mapping from the boot button where the IO could be, but still no success.

In other news: I tried disconnecting IO12 and the error message changed:
```
E (1209) sdmmc_cmd: sdmmc_card_init: send_if_cond (1) returned 0x109
E (1209) example: Failed to initialize the card (265). Make sure SD card lines have pull-up resistors in place.
```

negativekelvin, 2017-07-18:
If you can program you board the auto reset circuit is working fine, you don't need to worry about gpio 0 and 2.  You do need to worry about 12, which has to be connected after boot and before SD card init so put a nice delay in your program.

Staubgeborener, 2017-07-18:
Tried this workaround, but just in the moment when i plug in the wire the message suddenly appears again:

> E (7212) sdmmc_cmd: sdmmc_card_init: send_if_cond (1) returned 0x109
> E (7212) example: Failed to initialize the card (265). Make sure SD card lines have pull-up resistors in place.

llewellynnu, 2017-07-19:
Run the code in one-bit mode. This mode does not need IO12 to be connected.

Staubgeborener, 2017-07-19:
Tried all of the tipps now in 4-line and 1-line mode. The error messages will be the same, no difference.

I found the [IO0 Pin under the ESP32 Module](http://i.imgur.com/E0gHrcX.png). I could try to connect this one to IO2. Again: Do IO0 and IO2 only be connected while programming or can i solid a wire between these IOs?

llewellynnu, 2017-07-19:
Hi , 
If you are able to program the device then it cannot be an IO2 issue. 
This looks like a hardware fault please check your connections to the SD card

Staubgeborener, 2017-07-19:
Ok. I checked everything and it seems ok. But just in case and exclude anything, [here](http://i.imgur.com/D3qrlZC.jpg) is a (bad taken) picture of my connection.  I have the whole time a feeling, that the 10k pull up resistors could be the fault.

llewellynnu, 2017-07-19:
May I ask whats the transistor for in the image you have sent?

Staubgeborener, 2017-07-19:
It's not connected to my board in this case. I use this for sending a 'trigger signal' to push the load for some ms, so i keep my powerbank alive.  But it's now **not** connected.

Staubgeborener, 2017-07-19:
Well, i found, that i destroyed somewhere in this workaround my micro sd card (not the adapter). I can't be recognized from any device. So i changed it, but still no success.
Everytime i try to flash the esp32 (**with** connected IO12), i get the following error message:

> $ make flash monitor
> Flashing binaries to serial port COM3 (app at offset 0x10000)...
> esptool.py v2.0-beta3
> Connecting........_
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
> Flash params set to 0x0220
> Compressed 15824 bytes to 9216...
> Wrote 15824 bytes (9216 compressed) at 0x00001000 in 0.8 seconds (effective 154.6 kbit/s)...
> 
> A fatal error occurred: Timed out waiting for packet header
> make: *** [/home/Staubgeborener/esp/esp-idf/components/esptool_py/Makefile.projbuild:53: flash] Fehler 2

When i just do `make monitor`, there is always repeating

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> Falling back to built-in command interpreter.
> OK
> >ets Jun  8 2016 00:22:57

Now i disconnect IO12 from the sd card adapter and again `make flash monitor`: 

> $ make flash monitor
> Flashing binaries to serial port COM3 (app at offset 0x10000)...
> esptool.py v2.0-beta3
> Connecting........_
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
> Flash params set to 0x0220
> Compressed 15824 bytes to 9216...
> Wrote 15824 bytes (9216 compressed) at 0x00001000 in 0.8 seconds (effective 154.6 kbit/s)...
> 
> A fatal error occurred: Timed out waiting for packet header
> make: *** [/home/Staubgeorener/esp/esp-idf/components/esptool_py/Makefile.projbuild:53: flash] Fehler 2

Now i do `make monitor`,  again **without** IO12 connected:

> E (1218) sdmmc_cmd: sdmmc_card_init: send_scr returned 0x109
> E (1218) example: Failed to initialize the card (265). Make sure SD card lines have pull-up resistors in place.

Staubgeborener, 2017-07-20:
Can anyone, who has successfully done this, post his circuit and a 'real' picture of his setup?

Staubgeborener, 2017-07-23:
Just in case, maybe this is the fault: Is the Script only for SDMMC? So SDHC won't work? I use a SD/SDHC/SDXC Card Adapter with a SDHC Micro SD Card (8GB SanDisk).

Staubgeborener, 2017-07-23:
@llewellynnu @negativekelvin 
Just in case, maybe this is the fault: Is the Script only for SDMMC? So SDHC won't work? I use a SD/SDHC/SDXC Card Adapter (http://www.ebay.de/itm/400833592396) with a SDHC Micro SD Card (8GB SanDisk).

Staubgeborener, 2017-07-31:
To speak about this subject again: [Here](http://imgur.com/a/jQhWs) is a picture of my circuit made in firtzing. Not a beauty, but it shows perfectly my wiring. Is there a mistake? I think so - when i unplug my sd card i'll get the message at the beginning 'timeout waiting for packet handler'. When i plug it out, or better to say, when i plug out VCC (for example on breadboard line 23) it's 

> E (1934) sdmmc_cmd: sdmmc_card_init: send_op_cond (1) returned 0x107
> E (1934) example: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.

Alvin1Zhang, 2019-03-21:
@Staubgeborener Hi, sorry that get back to you so late, has your issue been resolved? Thanks.

closed 2019-04-08 by igrr