espressif/esp-idf #818: Flashing Issue -couldn't open port

Kartikkman, 2017-07-19:
`kartik@kartik-ThinkPad-Edge-E530 ~/esp/myapp $ make flash
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.0.1
Traceback (most recent call last):
  File "/home/kartik/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2412, in <module>
    _main()
  File "/home/kartik/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2405, in _main
    main()
  File "/home/kartik/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2134, in main
    esp = chip_class(args.port, initial_baud)
  File "/home/kartik/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 175, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/lib/python2.7/dist-packages/serial/__init__.py", line 85, in serial_for_url
    instance.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 294, in open
    raise SerialException(msg.errno, "could not open port %s: %s" % (self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
/home/kartik/esp/esp-idf/components/esptool_py/Makefile.projbuild:52: recipe for target 'flash' failed
make: *** [flash] Error 1
`

igrr, 2017-07-19:
Your user doesn't have access to /dev/ttyUSB0. You may try adding yourself to dialout group:

```
sudo usermod -a -G dialout $USER
```

Log out and then log back in after running this.

FayeY, 2017-07-20:
Hi @Kartikkman , did the above suggestion help you resolve the issue?

Kartikkman, 2017-07-20:
Hi , there was some connectivity problem b/w board & PC . 
It is now resolved. 

Thank you @FayeY , @iggr taking out your time to help me .

closed 2017-07-20 by Kartikkman

wevilp, 2018-10-06:
非常感谢，问题已经解决了

zuhaib446, 2018-12-24:
zuhai@zuhaib-hunzai MINGW32 ~/esp/esp-idf/examples/get-started/hello_world
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/zuhai/esp/esp-idf/requirements.txt are satisfied.
App "hello-world" version: v3.3-beta1-99-ge2fbee9f9
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.6-beta1
Serial port C:/msys32/dev/ttyUSB0
Traceback (most recent call last):
  File "C:/msys32/home/zuhai/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2865, in <module>
    _main()
  File "C:/msys32/home/zuhai/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2858, in _main
    main()
  File "C:/msys32/home/zuhai/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2565, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:/msys32/home/zuhai/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 213, in __init__
    self._port = serial.serial_for_url(port)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'C:/msys32/dev/ttyUSB0': WindowsError(2, 'The system cannot find the file specified.')
make: *** [/home/zuhai/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 1

zuhaib446, 2018-12-24:
`
zuhai@zuhaib-hunzai MINGW32 ~/esp/esp-idf/examples/get-started/hello_world
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/zuhai/esp/esp-idf/requirements.txt are satisfied.
App "hello-world" version: v3.3-beta1-99-ge2fbee9f9
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.6-beta1
Serial port C:/msys32/dev/ttyUSB0
Traceback (most recent call last):
  File "C:/msys32/home/zuhai/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2865, in <module>
    _main()
  File "C:/msys32/home/zuhai/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2858, in _main
    main()
  File "C:/msys32/home/zuhai/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2565, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:/msys32/home/zuhai/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 213, in __init__
    self._port = serial.serial_for_url(port)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'C:/msys32/dev/ttyUSB0': WindowsError(2, 'The system cannot find the file specified.')
make: *** [/home/zuhai/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 1

`

zuhaib446, 2018-12-24:
THANKS @KonstantinKondrashov

zuhaib446, 2018-12-24:
`$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/zuhai/esp/esp-idf/requirements.txt are satisfied.
App "hello-world" version: v3.3-beta1-99-ge2fbee9f9
Flashing binaries to serial port COM1 (app at offset 0x10000)...
esptool.py v2.6-beta1
Serial port COM1
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make: *** [/home/zuhai/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2
`

zuhaib446, 2018-12-24:
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/zuhai/esp/esp-idf/requirements.txt are satisfied.
App "hello-world" version: v3.3-beta1-99-ge2fbee9f9
Flashing binaries to serial port COM1 (app at offset 0x10000)...
esptool.py v2.6-beta1
Serial port COM1
Connecting......................................_____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make: *** [/home/zuhai/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2
[image]

henkiejan1, 2019-07-26:
For people who has also this isseu. Don´t forget to place a / before dev/ttyUSB0/ to it should be /dev/ttyUSB0/

sultan95678, 2019-08-27:
@igrr how can i log out .?
i know how log in .