espressif/esp-idf #819: Bluetooth Mesh support

OtherCrashOverride, 2017-07-19:
Support for the new Bluetooth Mesh standard needs to be addressed.

FHFS, 2017-07-19:
Do you mean this announcement by [SIG](https://www.bluetooth.com/news/pressreleases/2017/07/bluetooth-sig-announces-mesh-networking-capability) not more than 24 hours ago?

OtherCrashOverride, 2017-07-19:
Yes.  Nordic and others have a "day one" SDK available.

FHFS, 2017-07-19:
Does sound like a nice feature =)

TianHao-Espressif, 2017-07-21:
Thanks for your attention to ESP32. Standard Mesh Support is in plan, but the time need to discuss.

sthenc, 2017-07-27:
Hello, I am also interested in this. Any updates on the status?

Btw. the specification can be accessed publicly, I just downloaded it:
https://www.bluetooth.com/specifications/mesh-specifications

I fail to understand how big a difference there is between BLE mesh and normal BLE (but my understanding of both is very limited). 

I have a requirement for two BLE slave devices to exchange data between themselves.

@TianHao-Espressif  Do you think it would be possible to build a BLE mesh on top of the present esp32-bt-lib?
If implementing BLE mesh for ESP32 is not going to happen soon (meaning a couple of weeks), are there some already working alternatives?

FHFS, 2017-07-27:
@sthenc I think GATT server-client model would do for you.
[Mesh](https://en.wikipedia.org/wiki/Mesh_networking) is a network type, where you would be able to communicate with all devices on that network, even when for example that other node would be out of range.

jrewiens, 2017-09-01:
Is there any progress on ESP32 BLE mesh functionality?

TianHao-Espressif, 2017-09-25:
@jrewiens , for now, BLE -MESH is not ready in ESP32. You can consider about ESP-MESH with WIFI.

micheljung, 2017-09-26:
ESP-Mesh isn't ready for ESP32 either.

I'm willing to pay for BLE-Mesh support.

benjaminaigner, 2017-09-28:
@micheljung 
There is the BTSTack BLE stack, where they are currently developing BLE Mesh support.
I think @mringwal will be faster than the Bluedroid stack :-):

https://github.com/bluekitchen/btstack/tree/mesh

mringwal, 2017-09-28:
Bluetooth Mesh works by sending and receiving advertisements. I've seen that Espressif added support for scanning and advertising at the same time (well, it stops quickly to receive when sending). That's the basis for any Mesh implementation.

@benjaminaigner thanks for your confidence. We're fuzzy about the when, but we're working definitely on Bluetooth Mesh for BTstack (a few PTS tests are already green). It does have a few protocol layers to implement.

nvl1109, 2017-11-27:
Hello @TianHao-Espressif  and all,
Do you have any updates about BLE mesh for ESP32? Thank you.

Yulong-espressif, 2017-11-27:
@nvl1109 About the BLE Mesh. We are currently working on it and it will take time, so stay tuned for our follow-up IDF update. Thank you

mringwal, 2017-12-21:
BTstack's BLE Mesh supports Provisioning both as Device (new device that gets added to the network) and Provisioner (the device that provides network key to new device - similar to LE Secure Connections) in the test/mesh folder of the "mesh" branch. We plan to have basic network communication working maybe end of January.

Yulong-espressif, 2017-12-21:
@jo-nas We will support BLE Mesh at the first quarter of 2018. We are writing the code now. Thanks for your concerned.

NicksonYap, 2018-01-17:
@Yulong-espressif @TianHao-Espressif  
any news or beta release of BLE Mesh?

Yulong-espressif, 2018-01-22:
@NicksonYap We have now implemented some of the features of Bluetooth mesh. We plan to release the first version of the Bluetooth mesh in early April. Thank you for your interest in the progress of Bluetooth mesh.

wolfelec, 2018-02-22:
Hello Yulong-espressif -,

Will the bluetooth mesh be available with AT commands?

Thank you :)

wolfelec, 2018-02-26:
Hello @Yulong-espressif 

Will the bluetooth mesh be available with AT commands?

Thank you :)

Yulong-espressif, 2018-02-26:
@wolfelec We currently have no plans to join the Mesh AT command, we will consider other BLE Mesh release, thank you.

wolfelec, 2018-02-26:
@Yulong-espressif 

Thank you for your answer :)

So if I understand correctly, the new functionality expected for April, will not bring AT specific command to BLE Mesh?

Thank you

herry13, 2018-03-21:
Pin this thread because I'm very interested to use BLE mesh for my project.

micheljung, 2018-04-07:
@Yulong-espressif how's it going, are we on track? Early April ends soon :-)

rapzak, 2018-04-16:
@Yulong-espressif,
Hi - I am also interested in this :) :)

chegewara, 2018-04-16:
Few more interested people on this and i will create account on kickstart and write it by myself 🥇

sh4kesbeer, 2018-04-16:
Have been waiting to test BLE Mesh for two years (since the earliest rumors) now, would be great if you could bring it to the esp!

Yulong-espressif, 2018-04-16:
@sh4kesbeer @rapzak @micheljung Sorry for the delay, The BLE mesh code already finish now, we still need some time to test it, we will release it as soon as possible to the IDF. Thank you very much.

ZeeLivermorium, 2018-04-16:
Thank god! Y'all da best!

wolfelec, 2018-04-16:
@Yulong-espressif , thank you for the new :)

mik00, 2018-04-19:
Looking forward to BLE mesh release too.

manikandan-enability, 2018-05-09:
Any Update about BLE mesh for ESP32?
Thank you. 
@Yulong-espressif

leodesigner, 2018-05-09:
Maybe http://zephyrproject.org/ will be a faster solution for BLE Mesh ?

micheljung, 2018-05-09:
Last time I checked, they didn't even have bluetooth drivers fro ESP32 yet. I paid $100 via Bountysource so someone implemented it, but IIRC there were many other shortcomings as well so I gave up on Zephyr. @leodesigner

Shannon-I-am, 2018-05-22:
How is that testing going?
Can we expect BLE mesh to work with the existing wifi mesh interfaces?
I'm anxious to have this functionality.

chegewara, 2018-05-24:
I see new bluetooth option, a lot bluetooth options, one is suggesting we are having or we are very close to have implemented ble mesh. Some confirmation? @Weijian-Espressif @Yulong-espressif

micheljung, 2018-06-06:
@Yulong-espressif first you announced March, then you moved it to "early April", then you said you were "done" and now it's JUNE and we're still not there? I'm trying to run a business and I've been delaying my product waiting for BLE Mesh, if I knew these anouncements were so unreliable, I would've looked for another solution.

This really hurts Espressif's reputation. And my business.

Yulong-espressif, 2018-06-06:
@micheljung I'm sorry about the delay,  we have now completed the development of the BLE Mesh code protocol part. We can only provide to some customers at present. If you are in urgent need, please contact our business department. We can give priority to you. Feel sorry again.

Shannon-I-am, 2018-06-06:
@Yulong-espressif, will BLE Mesh ever be available without a 'special arrangement'? Maybe we should start looking elsewhere immediately?

denyip, 2018-06-08:
Hi, Is it possible to make a ESP32 as a bluetooth 5 device provisioner?

sh4kesbeer, 2018-06-08:
@OtherCrashOverride well, they have no obligation whatsoever to fulfil your demands, maybe you should be a bit more grateful for the fact that they are listening at all?

NicksonYap, 2018-06-08:
Let's keep this simple.
As @sh4kesbeer said, they have no obligation
A GitHub comment is not an official release date
It's our own responsibility to keep a look out at alternatives
This is a low cost platform, there's nothing much to expect

Yulong-espressif, 2018-06-08:
@all Thank you for your continuing interest in ESP32 BLE Mesh. At the moment we have decided internally that if there is no problem with the test, we will publish the BLE Mesh function to Github at the end of July.

wolfelec, 2018-06-08:
@Yulong-espressif Good news :)

OtherCrashOverride, 2018-06-08:
@sh4kesbeer 

> well, they have no obligation whatsoever to fulfil your demands

It was not a demand.  It was a request from a **customer**.  No threats were made.  No ultimatums were given (by me). Almost a year after my first inquiry, I have followed up on it.

> maybe you should be a bit more grateful for the fact that they are listening at all?

When a customer asks a question to a business, it is called **customer service**.  The level of customer service is also considered a very important measure of a company by many.  

As expressed earlier in the conversation, customers have their own businesses to run and their own schedules.  It is common practice in the industry to inquire about product "roadmaps".  If a vendor is not **reasonably** forthcoming about their "roadmaps" or chastises their customers for inquiring, then a customer may seek a relationship with a different vendor.  Its not an emotional reaction; rather, its simply common business practice in a highly competitive and time sensitive market.

Modern development is increasingly decentralized (global).  In this specific instance, it far simpler to select an existing BT Mesh solution than to work within the constraints of a 'special arrangement'.  When all development interaction has to go through 'approved channels', it significantly adds to the time and resources required for product development.

denyip, 2018-06-15:
@Yulong-espressif , do you mean that the release on Jul will be compatible to Bluetooth 5 mesh, that means it can communicate with other brand chips via Bluetooth 5 mesh? If it's true, then will be great.

Yulong-espressif, 2018-06-15:
@denyip ESP32 currently does not support the Bluetooth 5.0 feature, but BLE Mesh and 5.0 are independent. This means that our BLE Mesh is compatible with BLE Mesh on other chips, so feel free to use it. Thank you very much.

H0bbes, 2018-06-25:
Thank you for your hard work. Can't wait to start playing with it.
Did you solved/optimized the cohabitation between BLE scan/advertising and Wifi ?

denyip, 2018-06-25:
yes, it's basically, the system connect to wifi aws and then do scan ble simultaneously, run few months without problem.

ramams, 2018-07-25:
@Yulong-espressif is ESP BLE Mesh released? From your previous message it was expected to be released by end of July. Thanks in advance.

ngonhatnhi, 2018-08-02:
@Yulong-espressif Can you tell us released plan of ESP BLE Mesh ? We had to wait in very long time...

Shannon-I-am, 2018-08-02:
@ramams and @ngonhatnhi, I guess there was a problem with the test.

ngonhatnhi, 2018-08-03:
I will attempt to contact Espressif biz service, hope that we 'll get some news about BLE Mesh

Yulong-espressif, 2018-08-03:
Hi, @all, Thank you for your continued attention to BLE Mesh. At present, ESP32's BLE Mesh has been opened to the second batch of customers for testing. Due to business considerations, we are not yet able to publicly release the BLE Mesh version. If you need it urgently, please contact our business department. I am sorry for the inconvenience caused to you.

ngonhatnhi, 2018-08-03:
@Yulong-espressif Hello Mr.Yulong, i can not find email and phone number of your biz deparment in espressif.com , so can you help me ?

Shannon-I-am, 2018-08-03:
With such lack of clarity and inability to keep to schedules I'm switching from Espressif to Nordic.

jo-nas, 2018-08-30:
What about this? https://github.com/espressif/esp-mdf

Shannon-I-am, 2018-08-30:
@jo-nas that uses wifi, so although it may be relatively low power it's not going to be able to provide the battery-life of a BLE Mesh network.

ReanimationXP, 2018-09-04:
@Yulong-espressif any update on public release?

toxuin, 2018-10-02:
I think we should bump this more often.

sthenc, 2018-10-04:
This is taking so long I completely lost interest in this. 

With this not being done more than a year after the initial plans (and no one reverse-engineering the thing), I think it might even be possible they had some sort of silicon bug. Priority customers usually get the engineering samples first for the new silicon, and this is probably how long it takes to fix that sort of bug.

But I am not sure which is more disturbing, complete lack of transparency regarding that, or restricting access to basic functionality that the competitors have had for a year or so now.

[37 more replies]

closed 2020-07-27 by Alvin1Zhang