espressif/esp-idf #820: Feature request: FreeRTOS.TimerAPI.vTimerSetTimerID

samsam4, 2017-07-19:
[vTimerSetTimerID](http://www.freertos.org/FreeRTOS-timers-vTimerSetTimerID.html) is quite usefull function. Don't know by what reason is not included so far in the esp-idf distribution, but is quite good to have it.
Thanks

HubbyGitter, 2017-07-20:
On the other hand, it doesn't add anything which cannot be achieved without it, the worst case resulting in an additional pointer indirection, say, twice per timer expiration and an additionally stored pointer. In many cases, though, the pointer will refer to a small timer admin data structure anyway which will reside in the same memory location during runtime. On the downside of making this function available, the footprint of the IDF would increased.

Spritetm, 2017-07-20:
To be fair, if you do not use this function, from what I remember it'll not be linked into your final binary and should not incur any overhead.

HubbyGitter, 2017-07-20:
@Spritetm Ah yes... [`-ffunction-sections`, `-fdata-sections`](https://github.com/espressif/esp-idf/blob/master/make/project.mk#L235) and [`--gc-sections`](https://github.com/espressif/esp-idf/blob/master/make/project.mk#L201) are all there and [will do the trick](https://stackoverflow.com/questions/4274804/query-on-ffunction-section-fdata-sections-options-of-gcc)... thank you for pointing this out, and please disregard this aspect of my comment.

Dazza0, 2017-11-24:
@samsam4  `vTimerSetTimerID`  has been backported along with a number of other features from FreeRTOS v9.0.0 in commit 02304ad83e0a5f4815789d581446fa3afdd017b9 . Please see [this link](https://esp-idf.readthedocs.io/en/latest/api-guides/freertos-smp.html#backported-features) for more details.

closed 2018-01-08 by igrr