espressif/esp-idf #821: bluetooth scan issue (phy init data)

denyip, 2017-07-20:
I have another brand BLE SoC which act as a gatt server to adv it's device name say 'ABC'. When I use the ESP gatt client example to scan this ABC, it can find it and connect it. After some times, say few weeks, it cannot scan this 'ABC', but can scan other gatt server simulate by my smartphone, on the other hand, my smartphone can scan this 'ABC' and connect it.  I changed to another brand ESP32 module and the problem gone, I initally thin that it's the quality issue of the module, but somehow after few weeks, the problem happen again in the another brand module. I tried many ways to resolve this issue, one of the observation I found is that if I unclick the "Do phy calibration and store calibration data in NVS", the problem gone, but of course it showed some error say cannot save ble config something like that but at least it can scan the 'ABC' again and connect it. Then I think it's because the phy rf data stored in the nvs has broken, I tried choose the "Do phy calibration and store calibration data in NVS" optiotn againm and then 'make erase_flash' to completely erase the flash first and then flash the program to it, the problem gone. Is there anything I did wrong or it's  bug?

denyip, 2017-07-20:
is it the default setting in the gatt client example is require device bond? it seems if I set the setting to no bond, and then erase all flash and run again, the problem has gone, but if I set the security setting to bond, the problem appear again after first time connected, and then it will never discover the connected device ID.

denyip, 2017-07-24:
more info, I found that after below event occur, the "filter" will be created, and then I need to erase the entire flash include the phy init data partition data to resolve it.

I (720) GATTC_DEMO: Connect to the remote device.
--
I (720) GATTC_DEMO: 51 ae b0 70 4c e7
I (730) GATTC_DEMO: Searched Adv Data Len 23, Scan Response Len 0
I (730) GATTC_DEMO: Searched Device Name Len 0
I (740) GATTC_DEMO:
I (750) GATTC_DEMO: Stop scan successfully
I (830) GATTC_DEMO: Physical connection completed: ESP_GATTC_CONNECT_EVT: conn_id=0, gatt_if = 3, status =0
I (830) GATTC_DEMO: REMOTE BDA:
I (830) GATTC_DEMO: 00 02 5b 00 15 10
I (830) GATTC_DEMO: ESP_GAP_BLE_SEC_REQ_EVT=>esp_ble_gap_security_rsp
E (840) BT: lmp_version_below LMP version 6 < 8
I (850) GATTC_DEMO: Virtual connection completed: ESP_GATTC_OPEN_EVT conn_id 0, if 3, status 0, mtu 23
I (860) GATTC_DEMO: REMOTE BDA:
I (860) GATTC_DEMO: 00 02 5b 00 15 10
E (860) BT: l2cble_start_conn_update, the last connection update command still pending.
E (870) BT: l2cble_start_conn_update, the last connection update command still pending.
E (880) BT: l2cble_start_conn_update, the last connection update command still pending.
E (1220) BT: l2cble_start_conn_update, staus = 1, line = 509
E (1260) BT: btc_gattc_cb_handler: Unhandled event (17)!
I (1270) GATTC_DEMO: ESP_GAP_BLE_KEY_EVT
I (1270) GATTC_DEMO: key type = ESP_LE_KEY_PENC
I (1280) GATTC_DEMO: ESP_GAP_BLE_KEY_EVT
I (1280) GATTC_DEMO: key type = ESP_LE_KEY_PID
I (1280) GATTC_DEMO: ESP_GAP_BLE_KEY_EVT
I (1280) GATTC_DEMO: key type = ESP_LE_KEY_LID
D (1290) nvs: nvs_open bt_config.conf 1
D (1290) nvs: nvs_set_blob bt_cfg_key 250
D (1400) nvs: nvs_close 8
D (1400) nvs: nvs_open bt_config.conf 1
D (1400) nvs: nvs_set_blob bt_cfg_key 333
D (1400) nvs: nvs_close 9
D (1400) nvs: nvs_open bt_config.conf 1
D (1400) nvs: nvs_set_blob bt_cfg_key 393
D (1410) nvs: nvs_close 10
D (1410) nvs: nvs_open bt_config.conf 1
D (1410) nvs: nvs_set_blob bt_cfg_key 407
D (1420) nvs: nvs_close 11
I (1420) GATTC_DEMO: remote BD_ADDR: 00025b001510
I (1430) GATTC_DEMO: address type = 0
I (1430) GATTC_DEMO: pair status = fail
I (1610) GATTC_DEMO: SEARCH RES: conn_id = 0
I (1610) GATTC_DEMO: UUID16: 1800
I (1610) GATTC_DEMO: SEARCH RES: conn_id = 0
I (1610) GATTC_DEMO: UUID16: 1801
I (1610) GATTC_DEMO: SEARCH RES: conn_id = 0
I (1620) GATTC_DEMO: UUID16: 180a
I (1620) GATTC_DEMO: SEARCH RES: conn_id = 0
I (1630) GATTC_DEMO: UUID16: 180f
I (1630) GATTC_DEMO: SEARCH_CMPL: conn_id = 0, status 0

TianHao-Espressif, 2017-08-02:
@denyip , please try the release/v2.1 branch. We fix a bug of receive the packets from the device which has a bad RF performance.

denyip, 2017-09-12:
I cannot test it with the most updated branch because of the "node_list" issue in TW#15166

FayeY, 2017-10-10:
Any update? "node_list" issue has already fixed. Is this issue still unsolved?

closed 2017-10-16 by FayeY