espressif/esp-idf #823: SD MMC possible file IO memory leak

skateone, 2017-07-20:
Hi there. 
I have been using the SD MMC interface now for a while. it work well in 1 bit mode but something i have noticed is it consumes quite a bit of heap and doesnt release the memory upon file closes

from my alalysis it eats up 1.1kb of ram every time i perform a file write.
any Ideas?

pumelo, 2017-07-21:
Would you mind sharing the code which shows this behaviour?

skateone, 2017-07-22:
here we go... its not pretty and still a work in progress...
these routines allow me to move datafiles from an internal SPIFFS FS to the sd card for archiving.
also noticed that if i leave the card mounted permanently i end up with SD write errors as well so what i do is only mount the sd card when i need to perform a transfer. this helps for the mount disconnect issue..

as for the memory usage. i have tried modifying these routings to only update one file on the sd card compared to moving the files... and still the same response.. i think i have all my fcloses() in the right places. but yeah... what i do now is only perform an SD write every ten minutes or so to try reduce the heap usage. but end of the day other RTOS implementations i have used don't have this issue

also tried the arduino ESP32 libs for MMC. end of the day they utilize the below methods as well

char* concat(const char *s1, const char *s2) {
	char *result = (char *) malloc(strlen(s1) + strlen(s2) + 1); //+1 for the zero-terminator
	//in real code you would check for errors in malloc here
	strcpy(result, s1);
	strcat(result, s2);
	return result;
}

bool SDMount() {
	sdmmc_host_t host = SDMMC_HOST_DEFAULT();
	host.flags = SDMMC_HOST_FLAG_1BIT;
	sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
	slot_config.gpio_cd = (gpio_num_t) SD_DETECT;
	esp_vfs_fat_sdmmc_mount_config_t mount_config;
	mount_config.format_if_mount_failed = true;
	mount_config.max_files = 1;
	sdmmc_card_t* card;
	esp_err_t ret = esp_vfs_fat_sdmmc_mount(SDMOUNTPOINT, &host, &slot_config, &mount_config, &card);
	if (ret != ESP_OK) {
		if (ret == ESP_FAIL) {
			Serial.println("Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.");
			return false;
		} else {
			Serial.print("Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place.");
			Serial.println(ret, DEC);
			return false;
		}
		return false;
	}
	// Card has been initialized, print its properties
	sdmmc_card_print_info(stdout, card);

	///CMPLT
	SDCreateDir("/CMPLT");
	delay(200);
	return true;

}

bool SDUnmount() {
	if (esp_vfs_fat_sdmmc_unmount() == ESP_OK) {
		delay(200);
		return true;
	} else {
		return false;
	}
}

void SDFileRename(const char * source, const char * dest) {
	struct stat st;
	if (stat(concat("/sdcard/", dest), &st) == 0) {
		// Delete it if it exists
		unlink(concat("/sdcard/", dest));
	}
	rename(concat("/sdcard/", source), concat("/sdcard/", dest));
}

void SPIFFSListAndMoveReadyFiles() {
	int dd = 0;

	File root = SPIFFS.open("/");

	if (root) {
		File file = root.openNextFile();
		while (file) {
			if (!file.isDirectory()) {

				if (strchr(file.name(), 'U') != NULL) {
					//file found and ready to copy it to the sd card
					File readFile = SPIFFS.open(file.name());
					if (readFile) {
						Serial.println("U opened for read");
						Serial.println(file.name());
						bool error = false;

						FILE* sdFile = fopen(concat(SDMOUNTPOINT, file.name()), "w");
						if (sdFile != NULL) {
							while (readFile.available()) {

								uint32_t fSizeLeft = readFile.available();
								if (fSizeLeft > 1024) {
									char *readbuf = (char *) malloc(1024);
									int readCount = readFile.readBytes(readbuf, 1024);
									if (readCount == 1024) {
										//write to SD CARD
										fwrite(readbuf, 1024, 1, sdFile);
										//fputs(readbuf, sdFile);

										//SDFile.write((const unsigned char *) readbuf, 1024);
									} else {
										Serial.println("Not all bytes read");
										error = true;
										break;
										//do something here to terminate the process.
									}
									free(readbuf);
								} else {
									//read remainder bytes less the 1024.
									char *readbuf = (char *) malloc(fSizeLeft);
									int readCount = readFile.readBytes(readbuf, fSizeLeft);
									if (readCount == fSizeLeft) {
										//write to SD CARD
										//fputs(readbuf, sdFile);
										fwrite(readbuf, fSizeLeft, 1, sdFile);
										//SDFile.write((const unsigned char *) readbuf, fSizeLeft);
									} else {
										Serial.println("Not all bytes read");
										error = true;
										break;
										//do something here to terminate the process.
									}
									free(readbuf);
								}

							}

						} else {
							Serial.print("failed to open file : [");
							Serial.print(concat(SDMOUNTPOINT, file.name()));
							Serial.print("] for Writing");
						}
						fclose(sdFile);

						readFile.close();
						//SDFile.close();
						if (!error) {
							//perform SPIFFS Delete
							SPIFFSdeleteFile(file.name());
						}
					}
				}
			}
			file = root.openNextFile();
		}
		file.close();
	}
	root.close();
	delay(100);
	//globalConfigData.SDWriteINProgress = false;
}

negativekelvin, 2017-07-23:
"Not all bytes read" code path will skip freeing readbuf, but dont know if you are getting that error

igrr, 2017-07-23:
`concat`, allocates memory from the heap. Then you feed this pointer to fopen/stat/unlink and never `free` it.

But in general, can you reproduce the issue with sdmmc driver without using Arduino or SPIFFS?

skateone, 2017-07-23:
will make the relevant changes and report back thanks..

skateone, 2017-07-23:
constant heap usage of 152 bytes  now with the concat char freeing...  digging through my novice code now...

skateone, 2017-07-23:
testing going well... i was loosing my heap due to me attempting to use that new SPIFFS arduino lib.

i changed all the code to mount an internal VFS FAT partition and use that instead... so far the free heap has stabilized... and looking good... will report back a little later

igrr, 2017-10-13:
@skateone Do you consider this issue fixed?

closed 2017-11-21 by FayeY