espressif/esp-idf #824: Deep sleep wakeup does not work through a specific legal pin

husigeza, 2017-07-20:
Hello,

I cannot wake up the esp32 device from deep sleep through GPIO Pin26 (= RTC Pin7). It works with all other "legal" pins. Any ideas ?

Thanks!

jmorsap, 2017-09-12:
I'm having a possibly related issue. I've been able to get ULP wakeup from deepsleep to work on every pin i've tried, but cannot get the edge counter in the 'ULP' example to work on any pin but gpio 0/rtc_gpio 11. Is anyone else having trouble with this?

husigeza, I had trouble with some pins in the deep sleep example because I was leaving the pins floating. Are you using external pulldowns/pullups or using the internal ones like in the 'ULP example' ?'
`    gpio_num_t gpio_num = GPIO_NUM_0;
    rtc_gpio_set_direction(gpio_num, RTC_GPIO_MODE_INPUT_ONLY);
    rtc_gpio_pulldown_en(gpio_num);
    rtc_gpio_pullup_dis(gpio_num);
    rtc_gpio_hold_en(gpio_num);`
`

igrr, 2017-10-13:
@husigeza Thanks for reporting, this has been fixed in https://github.com/espressif/esp-idf/commit/64d1f947ed630fcdc0ef142493a67849e15b55e9.

closed 2017-10-13 by igrr