espressif/esp-idf #826: BLE Central: Can't connect to peripheral while scanning.

lucazader, 2017-07-20:
In my application I am developing the ESP32 so that as a ble central it can connect to multiple peripherals.
Ideally the app would be able to scan for devices and connect to them as the app discovers them, without stopping the scan.

However I have come across an issue where the app wont connect while scanning.
The app is performing a BLE scan.
The app then tries to connect to a discovered peripheral, while still scanning. The ESP_GATTC_OPEN_EVT returns either "ESP_GATT_ERROR" or "ESP_GATT_BUSY". 

If the app stops scanning, then tries to connect, the ESP_GATTC_OPEN_EVT returns with "ESP_GATT_OK"

Please let me know if any more information is needed.

lucazader, 2017-07-27:
Is anyone able to comment any further on this one?

@igrr @wujiangang ?

heyinling, 2017-07-30:
@lucazader The following figure is from BLE core Spec v4.2, Vol3 part C Section 9.3.6. it's a reasonable implementation that can't connect when scanning.

[image]

negativekelvin, 2017-07-30:
Sure it is _reasonable_, but we want super turbo esp32.  We want to use every possible radio time slice to do something useful whether it is scan, advertise, rx, tx, etc.  And that has to be optimized at the lower level.  Of course you can't do everything at once, but you can try...

lucazader, 2017-09-20:
have created work around to stop scanning, then connect. Will close issue for now

closed 2017-09-20 by lucazader