espressif/esp-idf #827: WIFI calibration cycle drawing more current than expected

skateone, 2017-07-21:
HI THERE...
after i updated to the latest code i am getting brown out detections on boot.
I have narrowed it down to the WIFI calibration at is drawing more current than expected.
can anyone tell me why this has changed and how do we rectify it as my design doesnt cater for more than 400mA draw on the 3.3v supply

igrr, 2017-07-21:
Calibration behavior has not changed for a fairly long time. However brownout detection has been recently enabled in ESP-IDF by default (previously it wasn't implemented). If you see brownouts happening on one of our dev kits, please mention which board that is. If you see this on custom hardware, please check if it meets hardware design guidelines (you can find this document in espressif.com docs section).

negativekelvin, 2017-07-22:
I don't think there is good information on calibration: current draw ranges for the 3 modes, possible phy performance effects when no calibration, recommended conditions for factory calibration.

```
PHY_RF_CAL_PARTIAL = 0x00000000,        /*!< Do part of RF calibration. This should be used after power-on reset. */
    PHY_RF_CAL_NONE    = 0x00000001,        /*!< Don't do any RF calibration. This mode is only suggested to be used after deep sleep reset. */
    PHY_RF_CAL_FULL  
```

skateone, 2017-07-22:
hi there.. i found my brown out problem.. i am using custom hardware that i designed.. the issue was a dry joint on a boost cap for the main PSU module... go figure.. the smallest thing causing the most hassle.. out of 100 devices it had to be my test unit with this failure...
thank you for the responses.  the one thing i did also do was to re-enable the Phy-init partition for the radio calibration params. i see now it only does full calibration on first boot after erasing all the flash.

closed 2017-07-23 by igrr