espressif/esp-idf #829: millis() gives multiple of 10 msec

vortex314, 2017-07-21:
While porting code from ESP8266 to ESP32-IDF , I got the impression that the function millis() only has a resolution of 10 msec. 
Since I need a precision of 1 msec, what other function should I use ?
void loop() {
  for(int i=0;i<20;i++) {
    delay(1);
    Serial.printf("%ld \r\n",::millis());
  }
}

devsaurus, 2017-07-21:
This might be caused by the standard tick rate of 100 Hz in FreeRTOS, just speculating. Will millis() improve when you set the tick rate to 1000 in menuconfig?

vortex314, 2017-07-21:
Indeed looks to be that way. I didn't test it as I took micros() to create a 64 bit counter. 
But the code of millis() => xTaskGetTickCount() * portTICK_PERIOD_MS
#define portTICK_PERIOD_MS			( ( TickType_t ) 1000 / configTICK_RATE_HZ )
#define configTICK_RATE_HZ				( CONFIG_FREERTOS_HZ )
#define CONFIG_FREERTOS_HZ 100
Indeed there are 100 ticks per second , and it's multplied by 10 for having the milliseconds.
Wonder why the default is only 100Hz ? 
You can put this at 1000 in menuconfig, then probably millis() will be precise in msec.

closed 2017-07-21 by vortex314