espressif/esp-idf #832: bluetooth connect issue (device disconnect)

denyip, 2017-07-24:
I use the gattc example to connect another ble SoC, and got below result, it seems the pairing is success, serivce and char can be discovered but it will disconnect after around 15 second. I have checked the SoC program that there is no Idle timeout timer to disconnect, also I use but android phone to connect, it won't discconnect.

I (5500) GATTC_DEMO: Searched Adv Data Len 21, Scan Response Len 0
I (5510) GATTC_DEMO: Searched Device Name Len 3
I (5510) GATTC_DEMO: ESS
I (5510) GATTC_DEMO: 

I (5520) GATTC_DEMO: Searched device ESS

I (5520) GATTC_DEMO: Connect to the remote device.
I (5530) GATTC_DEMO: Stop scan successfully
I (5930) GATTC_DEMO: Physical connection completed: ESP_GATTC_CONNECT_EVT: conn_id=0, gatt_if = 3, status =0
I (5930) GATTC_DEMO: REMOTE BDA:
I (5930) GATTC_DEMO: 00 02 5b 00 15 10 
I (5940) GATTC_DEMO: ESP_GAP_BLE_SEC_REQ_EVT=>esp_ble_gap_security_rsp
E (5950) BT: Non bonding: No keys will be exchanged
E (5950) BT: lmp_version_below LMP version 6 < 8
I (5960) GATTC_DEMO: Virtual connection completed: ESP_GATTC_OPEN_EVT conn_id 0, if 3, status 0, mtu 23
I (5970) GATTC_DEMO: REMOTE BDA:
I (5970) GATTC_DEMO: 00 02 5b 00 15 10 
E (5980) BT: l2cble_start_conn_update, the last connection update command still pending.
E (5980) BT: l2cble_start_conn_update, the last connection update command still pending.
E (5990) BT: l2cble_start_conn_update, the last connection update command still pending.
I (6010) GATTC_DEMO: remote BD_ADDR: 00025b001510
I (6010) GATTC_DEMO: address type = 0
I (6010) GATTC_DEMO: pair status = success
E (6330) BT: l2cble_start_conn_update, staus = 1, line = 509
I (6530) GATTC_DEMO: SEARCH RES: conn_id = 0
I (6530) GATTC_DEMO: UUID16: 1800
I (6530) GATTC_DEMO: SEARCH RES: conn_id = 0
I (6530) GATTC_DEMO: UUID16: 1801
I (6540) GATTC_DEMO: SEARCH RES: conn_id = 0
I (6540) GATTC_DEMO: UUID16: 180a
I (6550) GATTC_DEMO: SEARCH RES: conn_id = 0
I (6550) GATTC_DEMO: UUID16: 180f
I (6560) GATTC_DEMO: SEARCH_CMPL: conn_id = 0, status 0
E (36040) BT: btm_sec_clr_temp_auth_service() - no dev CB

E (36040) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013
E (36040) BT: Device not found

E (36040) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 00:02:5b:00:15:10
I (36050) GATTC_DEMO: Physical connection closed: ESP_GATTC_DISCONNECT_EVT
I (36060) GATTC_DEMO: Virtual connection closed: ESP_GATTC_CLOSE_EVT

TianHao-Espressif, 2017-08-02:
Could you tell me your ESP-IDF version (commit ID)

FayeY, 2017-08-09:
Hi @denyip , any update? Could you have a try with the latest ESP-IDF?

denyip, 2017-08-09:
sorry, where can i check the idf version? the commit id is 4ec2abb....

i simply disabled the key paring feature of my counter part SoC to avoid this problem, and continued the 

development of other areas. I will update the latest idf and try again later.

FayeY, 2017-09-05:
Hi @denyip , this issue solved now?

closed 2017-10-10 by FayeY

denyip, 2017-10-13:
It seems the gatt client api  of this idf version has large modification compare with the one I am using, I need to modified my code for this new version idf, no time to perform the test yet.