espressif/esp-idf #833: ESP32 fixed data rate

liulingliuf, 2017-07-24:
您好，我现在使用esp32WiFi芯片，需要固定发送数据速率。具体来说，我们想用固定的高速率传输文件。在ＥＳＰ８２６６中有一个函数叫做WiFi_set_user_fixed_rate，头文件中定义了原型。但是，在ESP32我们未找到相关函数，可否做一个临时的客户版本？

liuzfesp, 2017-07-24:
@liulingliuf, 相对于8266, ESP32的API有较大的调整，有一些API现在并不支持，或者还未对外开放。针对你说的这个要求，我争取这两天给你搞一个临时版本。　另外，　也许你可以将你的需求说得更具体点儿，　如，　是使用udp还是tcp或者别的协议进行传输？　传输的文件大概多大？　在什么样的场景下进行传输，信道条件较好还是较差，距离较远还是较近？　目前系统可用的内存为多少？　等等，　依据这些信息，可以向你推荐一组配置值。

liulingliuf, 2017-07-24:
用UDP协议发送，单个UDP包长4000-10000，越大越好。
信道条件好，距离短。
用的模组型号是：ESP32-WROOM-32，用户程序比较小。

liuzfesp, 2017-07-24:
UDP是不可靠协议，接收端会有丢包和重复包检测么？

liulingliuf, 2017-07-24:
UDP丢包和重复包在接收端会有检测。
最渴望的是能够把发送速率固定住，可控，自由设置。
其次，包长越长越好。
最好能够有像ESP8226那样有wifi_send_pkg_freedom这样在底层可以直接发包的程序。
谢谢！

liuzfesp, 2017-07-24:
Got it. 这两天我看一下。

liulingliuf, 2017-07-24:
多谢！等你好消息

liulingliuf, 2017-07-25:
@liuzfesp 昨天测试结束后，需要一个新的特性: 在wifi通过tcp/ip协议发送的时候，数据是经过加密的。这个加密是在tcp/ip协议栈内加密的还是在wifi driver上加密的?可否设置在指定的包上不经过加密直接发送？

liuzfesp, 2017-07-25:
@liulingliuf 是否加密取决于连接的ap. 
另外，还有几个问题需要确认一下， ESP32是用作AP还是STA, 或者一端作为AP, 一端作为STA？ 另外，你准备配置什么样的速率？ 这两个问题需要先确认，不然编出来的临时版本可能会有问题。

liulingliuf, 2017-07-25:
@liuzfesp
ESP32是作为AP使用，至少支持设置为54M。
这个加密是在wifi driver上进行的还是在lwip上进行的？

liuzfesp, 2017-07-25:
加密是在WiFi Driver上。

liulingliuf, 2017-07-25:
能否在指定包上disable掉?

liuzfesp, 2017-07-25:
如果disable掉的话代码改动很大，以前也没有偿试过这样做过。 如果不想AP加密，AP可以不设置密码。

liulingliuf, 2017-07-25:
好的，理解了，多谢！

liuzfesp, 2017-07-25:
目前fix rate功能基本可用，但不完善，而且在fix rate情况下ampdu未能够跑起来（ampdu没跑起来会很影响性能）。 你先试一下看看有什么问题。 代码见[https://github.com/liuzfesp/esp-idf/tree/customer/fix_rate_for_github_customer_request](url) ， 具体的使用方法见链接idf下的readme文件。 你先试一下，有问题再找我。

liuzfesp, 2017-07-25:
要注意，使用UDP发包的时候，需要判断返回值，如果返回值为no memory， 则需要做flow control， 否则会大量丢包。

liulingliuf, 2017-07-26:
@liuzfesp为什么代码不能看了

liuzfesp, 2017-07-26:
你是说链接打不开还是？

liuzfesp, 2017-07-26:
如果是链接打不开的话，可以将链接copy到浏览器的地址栏再打开试试。

liulingliuf, 2017-07-26:
不好意思唐突打扰你，直接点击是不能打开的，后来我在你的主页里找到了

FayeY, 2017-07-31:
@liulingliuf ，请问这个问题现在解决了吗？

closed 2017-08-07 by FayeY

liulingliuf, 2017-08-07:
@FayeY 已经解决，谢谢！