espressif/esp-idf #840: SDMMC cannot works on slot0

petolat, 2017-07-25:
Hello,

I'm trying to run the sd_card example on slot0 in 4 wires, HS1_* signal in the IO_MUX. But the esp32 freeze... I'm working on the latest eps-idf (cloned today, so the 3.0).
I have the pull up on the lines.
I cannot use the slot1 because I need to use the JTAG port in the same time of sdcard.

In sd_car_example_main.c, I only modify  those lines:

**void app_main(void)
{
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.slot = SDMMC_HOST_SLOT_0;

    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
    slot_config.width =4;
    ...**

It freeze in **configure_pin(pslot->clk);** (esp-idf/components/driver/sdmmc_host.c, line 325) called in 
esp_err_t err = sdmmc_host_init_slot(host_config->slot, slot_config); (esp-idf/components/fatfs/src/vfs_fat_sdmmc.c, line 60)

How should I do ?

negativekelvin, 2017-07-25:
HS1 conflict with flash pins SPI0 so doesn't work

igrr, 2017-07-26:
As @negativekelvin has pointed out, you can't use the SD card and SPI flash on the same set of pins, because the two protocols are not compatible. To use slot 0, you need to connect SPI flash to a different set of pins, and program the numbers of the pins into efuses.
There is also an SD card driver for SPI which has been merged recently and should appear on Github soon. It lets you connect the SD card to arbitrary pins, with ~4x slower data rate compared to SD interface.

closed 2017-07-26 by igrr