espressif/esp-idf #844: http_request example hangs

ujeenator, 2017-07-27:
http_request example hangs on this line:
r = read(s, recv_buf, sizeof(recv_buf)-1);

after about 10 hours of workings

idf version: v2.1-rc1

krzychb, 2017-07-27:
- Is it repeatable, behaviour i.e. did you test it couple of times and it always hangs on the same line after about 10 hours of working?
- How did you test it hangs on this particular line?

ujeenator, 2017-07-27:
>> Is it repeatable, behaviour i.e. did you test it couple
>> of times and it always hangs on the same 

Yes, it's repeatable on random intervals of time, when I modified example to make requests every 2 secs it start to hang every hour

Look like this example just missing timeout setup for data receiving, I've added this code:
```
struct timeval timeout;
timeout.tv_sec = 10;
timeout.tv_usec = 0;
if ( setsockopt(s, SOL_SOCKET, SO_RCVTIMEO, &timeout, sizeof(timeout)) < 0) {
    ESP_LOGE(TAG, "... failed to set socket read timeout");
    close(s);
    vTaskDelay(4000 / portTICK_PERIOD_MS);
    continue;
}
ESP_LOGI(TAG, "... set socket read timeout success");
```

 ....and run it on second board side by side with original code, the original one already hanged for 5 times, and new one keep working without problems

ujeenator, 2017-07-27:
>> How did you test it hangs on this particular line?

I just added more logging.

krzychb, 2017-07-28:
> Yes, it's repeatable on random intervals of time, when I modified example to make requests every 2 secs it start to hang every hour

Good catch. I see similar scenario. I have made two tests and for requests every 2 seconds, it hanged after about 70 and 80 min. 

> Look like this example just missing timeout setup for data receiving,

Did additional check for timeout definitively resolved this issue?

ujeenator, 2017-07-28:
> Did additional check for timeout definitively resolved this issue?

For now I made 24 hour non-stop test, with timeout setup all works perfectly

Also when checks read(...) func exit code for -1 we can tell when timeout happens

I will made more tests for next week and share result

FayeY, 2017-08-07:
Hi @ujeenator , any update? Is this issue solved now?

negativekelvin, 2017-08-08:
@FayeY solution proposed above but example code not updated and no pull request so issue should remain open

closed 2017-11-21 by FayeY