espressif/esp-idf #845: The Station connected to AP of ESP32 is periodically disconnected

kimdowan84, 2017-07-27:
Hello!

I am using two esp32s.
One is APSTA and the other is used as an STA that connects to it.
However, the STA connected to the AP will be disconnected with the following message.

----------------------------
I (621498) wifi: state: run -> auth (4a0)
I (621498) wifi: n:4 0, o:4 1, ap:4 1, sta:4 1, prof:4
     : STA_DISCONNECTED
I (625724) wifi: state: auth -> init (0)
----------------------------

And Reconnecting to AP, this happens every five minutes.

I heard it was due to WIFI_REASON_ASSOC_EXPIRE, is this solved?

Thank you.

liuzfesp, 2017-07-31:
@kimdowan84 I think this issue is already fixed, will be merge into IDF master recently, maybe this week.

FayeY, 2017-09-18:
Hi @kimdowan84 , have you tried the latest IDF? Is this issue solved?

kimdowan84, 2017-09-19:
At idf_version = v3.0-dev-317..., not solved.

Yes.
At idf_version = v3.0-dev-710..., it may be solved.

kimdowan84, 2017-09-20:
Hi FayeY!

I have tested with v3.0-dev-710... but there is  **BIG  problem.**

The client that is connected to AP(v3.0-dev-710) could not get IP address. 
Despite receiving the "SYSTEM_EVENT_STA_CONNECTED" event, 
I did not receive the "SYSTEM_EVENT_STA_GOT_IP" event.

To connect to other AP(v3.0-dev-317), I got the envent "SYSTEM_EVENT_STA_GOT_IP"

The same results were obtained using a smartphone.

TimXia, 2017-09-20:
@kimdowan84 I tested with v3.0-dev-717. It's OK. I received the "SYSTEM_EVENT_STA_GOT_IP" event. Smartphone is also OK.

kimdowan84, 2017-09-21:
Hi TimXia

Lastest IDF version is v3.0-dev-710, isn't it?

TimXia, 2017-09-22:
@kimdowan84 Sorry, the latest IDF has not updated to github yet. But it will be soon. Maybe the next week.

TimXia, 2017-09-25:
@kimdowan84 Sorry, it's a bug. We have found the root cause. I will be fixed these days.

kimdowan84, 2017-10-14:
Yes, 806 version is OK!

closed 2017-10-16 by FayeY