espressif/esp-idf #847: Creating ringbuffer failed

Staubgeborener, 2017-07-27:
I try to create a ringbuffer with this librarys
```
#include "freertos/FreeRTOS.h"
#include <freertos/queue.h>
#include "freertos/ringbuf.h"
```
but i'll get only error messages:

```
RingbufHandle_t packetRingbuf;
packetRingbuf=xRingbufferCreate(12*1024, RINGBUF_TYPE_NOSPLIT);[/code]
```

> $ make flash
> CXX promiscuous_mode_esp32.o
> D:/msys32/home/Staubgeborener/esp/esp-idf/promiscuous_mode_esp32/main/promiscuous_mode_esp32.cpp:36:1: error: 'packetRingbuf' does not name a type
>  packetRingbuf=xRingbufferCreate(12*1024, RINGBUF_TYPE_NOSPLIT);

```
RingbufHandle_t packetRingbuf = xRingbufferCreate(1024, RINGBUF_TYPE_NOSPLIT);
```

>  $ make flash
>     CXX promiscuous_mode_esp32.o
>     AR libmain.a
>     LD promiscuous_mode_esp32.elf
>     D:/msys32/home/Staubgeborener/esp/esp-idf/promiscuous_mode_esp32/build/main\libmain.a(promiscuous_mode_esp32.o):(.literal._Z41__static_initialization_and_destruction_0ii+0x18): undefined reference to `xRingbufferCreate(unsigned int, ringbuf_type_t)'
>     D:/msys32/home/Staubgeborener/esp/esp-idf/promiscuous_mode_esp32/build/main\libmain.a(promiscuous_mode_esp32.o): In function `__static_initialization_and_destruction_0(int, int)':
>     d:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:311: undefined reference to `xRingbufferCreate(unsigned int, ringbuf_type_t)'
>     collect2.exe: error: ld returned 1 exit status
>     make: *** [D:/msys32/home/Staubgeborener/esp/esp-idf/make/project.mk:323: /home/Staubgeborener/esp/esp-idf/promiscuous_mode_esp32/build/promiscuous_mode_esp32.elf] Fehler 1

```
RingbufHandle_t packetRingbuf;
xRingbufferCreate(1024, RINGBUF_TYPE_NOSPLIT);
```

>     $ make flash
>     CXX promiscuous_mode_esp32.o
>     D:/msys32/home/Staubgeborener/esp/esp-idf/promiscuous_mode_esp32/main/promiscuous_mode_esp32.cpp:36:18: error: expected constructor, destructor, or type conversion before '(' token
>     xRingbufferCreate(1024, RINGBUF_TYPE_NOSPLIT);
>     ^
>     make[1]: *** [/home/Staubgeborener/esp/esp-idf/make/component_wrapper.mk:203: promiscuous_mode_esp32.o] Fehler 1
>     make: *** [D:/msys32/home/Staubgeborener/esp/esp-idf/make/project.mk:386: main-build] Fehler 2

Spritetm, 2017-07-27:
Seems ringbuf.h doesn't have C++ guards around it. For now, you can work around it by doing
``
extern "C" {
#include "freertos/ringbuf.h"
}
``

Staubgeborener, 2017-07-27:
Great! Thank you so much, this seems to work:
```c++
#include "freertos/FreeRTOS.h"
#include <freertos/queue.h>
extern "C" {
#include "freertos/ringbuf.h"
}

RingbufHandle_t packetRingbuf=xRingbufferCreate(1024, RINGBUF_TYPE_NOSPLIT);
```

closed 2017-07-27 by Staubgeborener