espressif/esp-idf #849: after a new install: fatal: 'git status --porcelain=2'

ESP32DE, 2017-07-27:
whas there a change in anything?
> fatal: 'git status --porcelain=2' failed in submodule components/aws_iot/aws-iot-device-sdk-embedded-C

does one know what this shell line mean - which command is not found?

best wishes
rudi ;-)

FHFS, 2017-07-28:
remove the submodule then "git submodule update --init --force --remote --recursive"?

FayeY, 2017-07-31:
Hi @ESP32DE , did the above suggestion help you resolve the issue?

ESP32DE, 2017-07-31:
@FHFS 

txs i will try your suggestion. 
btw it is the [aws-iot-device-sdk-embedded-C](https://github.com/espressif/esp-idf/tree/master/components/aws_iot)
after `git clone --recursive https://github.com/espressif/esp-idf.git` 
and `git submodule update --init`
there is still `.git file`  in the folder
if i remove this file this solve the problem ( too ) 
but can't then update later cause .git file then is missing.

@FayeY 
i will check again my steps and will response here 
cause we have more extern repos in the esp-idf 
but only aws makes a problem, perhabs there is still a cmd missing
i will follow the update steps in detail now, what is missing.
thank you!

best wishes
rudi ;-)

ESP32DE, 2017-07-31:
@FHFS 
txs - the .git file is again then in the supmodule and disturb
( after delete this .git file - then there is no error ) 

btw we have more .git files in the submodule that are forked
but only this AWS makes a problem

example ( no problem with it ) 
[image]

best wishes
rudi ;-)

igrr, 2017-08-17:
Hi rudi, can you give the steps to reproduce this (once you are back from your vacation)?
Which tool do you use to clone? Tortoisegit, command line git in mingw, command line git in cygwin, git in virtual machine using shared folder, etc.
What steps do you do after cloning?
Thanks.

FayeY, 2017-11-21:
Any update about this issue?

closed 2017-12-01 by FayeY

pglen, 2018-11-23:
On MSYS2 the /etc/profile.d/esp_env* has an extra slash and quote. Editing them out fixes the issue.