espressif/esp-idf #850: mbedtls_ssl_handshake hangs

lenzarda, 2017-07-27:
Hi
I am using a modified version of the example https_request with esp-idf latest commit 843159b.
As recommended in #630 the "Enable hardware MPI (bignum) acceleration" is not enabled.
mbedtls_ssl_handshake works not very often. In the majority of the runs I get the following logs.
Any idea on how to fix this?
It is really blocking for me.
Thanks
Davide

--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:3736
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:11908
entry 0x40079404
I (1517) heap_init: Initializing. RAM available for dynamic allocation:
I (1518) heap_init: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
D (1538) heap_init: New heap initialised at 0x3ffae2a0
I (1586) heap_init: At 3FFB79A8 len 00028658 (161 KiB): DRAM
D (1644) heap_init: New heap initialised at 0x3ffb79a8
I (1692) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1752) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1812) heap_init: At 40094498 len 0000BB68 (46 KiB): IRAM
D (1869) heap_init: New heap initialised at 0x40094498
D (2033) heap_init: New heap initialised at 0x3ffe0440
D (2033) heap_init: New heap initialised at 0x3ffe4350
I (2033) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
D (2044) nvs: nvs_flash_init_custom start=9 count=6
D (2051) nvs: nvs_open misc 1
D (2051) nvs: nvs_get_str_or_blob log
I (2054) wifi: wifi firmware version: 9ffec4d
I (2058) wifi: config NVS flash: enabled
I (2062) wifi: config nano formating: disabled
D (2066) nvs: nvs_open nvs.net80211 1
D (2069) nvs: nvs_get opmode 1
D (2073) nvs: nvs_get country 1
D (2076) nvs: nvs_get_str_or_blob sta.ssid
D (2080) nvs: nvs_get_str_or_blob sta.mac
D (2084) nvs: nvs_get sta.authmode 1
D (2088) nvs: nvs_get_str_or_blob sta.pswd
D (2092) nvs: nvs_get_str_or_blob sta.pmk
D (2096) nvs: nvs_get sta.chan 1
D (2099) nvs: nvs_get auto.conn 1
D (2103) nvs: nvs_get bssid.set 1
D (2106) nvs: nvs_get_str_or_blob sta.bssid
D (2110) nvs: nvs_get sta.phym 1
D (2114) nvs: nvs_get sta.phybw 1
D (2117) nvs: nvs_get_str_or_blob sta.apsw
D (2121) nvs: nvs_get_str_or_blob sta.apinfo
D (2126) nvs: nvs_get_str_or_blob ap.ssid
D (2130) nvs: nvs_get_str_or_blob ap.mac
D (2134) nvs: nvs_get_str_or_blob ap.passwd
D (2138) nvs: nvs_get_str_or_blob ap.pmk
D (2142) nvs: nvs_get ap.chan 1
D (2145) nvs: nvs_get ap.authmode 1
D (2149) nvs: nvs_get ap.hidden 1
D (2152) nvs: nvs_get ap.max.conn 1
D (2155) nvs: nvs_get bcn.interval 2
D (2159) nvs: nvs_get ap.phym 1
[... 378 lines cut ...]
I (6323) mbedtls: ssl_tls.c:2449 message length: 410, out_left: 410

I (6332) mbedtls: ssl_tls.c:2455 ssl->f_send() returned 410 (-0xfffffe66)

I (6337) mbedtls: ssl_tls.c:2474 <= flush output

I (6343) mbedtls: ssl_tls.c:2864 <= write record

I (6348) mbedtls: ssl_cli.c:1051 <= write client hello

I (6354) mbedtls: ssl_cli.c:3254 client state: 2

I (6360) mbedtls: ssl_tls.c:2430 => flush output

I (6365) mbedtls: ssl_tls.c:2442 <= flush output

I (6371) mbedtls: ssl_cli.c:1403 => parse server hello

I (6377) mbedtls: ssl_tls.c:3738 => read record

I (6382) mbedtls: ssl_tls.c:2222 => fetch input

I (6388) mbedtls: ssl_tls.c:2380 in_left: 0, nb_want: 5

I (13690) wifi: pm start, type:0

lenzarda, 2017-07-27:
Here are the logs of a successful SSL negotiation
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:3736
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:11908
entry 0x40079404
I (1517) heap_init: Initializing. RAM available for dynamic allocation:
I (1518) heap_init: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
D (1538) heap_init: New heap initialised at 0x3ffae2a0
I (1586) heap_init: At 3FFB79A8 len 00028658 (161 KiB): DRAM
D (1644) heap_init: New heap initialised at 0x3ffb79a8
I (1692) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1752) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1812) heap_init: At 40094498 len 0000BB68 (46 KiB): IRAM
D (1869) heap_init: New heap initialised at 0x40094498
D (2033) heap_init: New heap initialised at 0x3ffe0440
D (2033) heap_init: New heap initialised at 0x3ffe4350
I (2033) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
D (2044) nvs: nvs_flash_init_custom start=9 count=6
D (2051) nvs: nvs_open misc 1
D (2051) nvs: nvs_get_str_or_blob log
I (2054) wifi: wifi firmware version: 9ffec4d
I (2058) wifi: config NVS flash: enabled
I (2062) wifi: config nano formating: disabled
D (2066) nvs: nvs_open nvs.net80211 1
D (2070) nvs: nvs_get opmode 1
D (2073) nvs: nvs_get country 1
D (2076) nvs: nvs_get_str_or_blob sta.ssid
D (2080) nvs: nvs_get_str_or_blob sta.mac
D (2084) nvs: nvs_get sta.authmode 1
D (2088) nvs: nvs_get_str_or_blob sta.pswd
D (2092) nvs: nvs_get_str_or_blob sta.pmk
D (2096) nvs: nvs_get sta.chan 1
D (2099) nvs: nvs_get auto.conn 1
D (2103) nvs: nvs_get bssid.set 1
D (2106) nvs: nvs_get_str_or_blob sta.bssid
D (2110) nvs: nvs_get sta.phym 1
D (2114) nvs: nvs_get sta.phybw 1
D (2117) nvs: nvs_get_str_or_blob sta.apsw
D (2121) nvs: nvs_get_str_or_blob sta.apinfo
D (2126) nvs: nvs_get_str_or_blob ap.ssid
D (2130) nvs: nvs_get_str_or_blob ap.mac
D (2134) nvs: nvs_get_str_or_blob ap.passwd
D (2138) nvs: nvs_get_str_or_blob ap.pmk
D (2142) nvs: nvs_get ap.chan 1
D (2145) nvs: nvs_get ap.authmode 1
D (2149) nvs: nvs_get ap.hidden 1
D (2152) nvs: nvs_get ap.max.conn 1
D (2156) nvs: nvs_get bcn.interval 2
D (2159) nvs: nvs_get ap.phym 1
D (2162) nvs: nvs_get ap.phybw 1
D (2166) nvs: nvs_get ap.sndchan 1
D (2170) nvs: nvs_set_blob sta.mac 6
D (2182) nvs: nvs_set_blob ap.mac 6
I (2183) wifi: Init dynamic tx buffer num: 32
I (2183) wifi: Init data frame dynamic rx buffer num: 64
I (2185) wifi: Init management frame dynamic rx buffer num: 64
I (2191) wifi: wifi driver task: 3ffc5260, prio:23, stack:4096
[... 474 lines cut ...]
I (7329) mbedtls: ssl_tls.c:2222 => fetch input

I (7335) mbedtls: ssl_tls.c:2380 in_left: 5, nb_want: 2481

I (7341) mbedtls: ssl_tls.c:2404 in_left: 5, nb_want: 2481

I (7348) mbedtls: ssl_tls.c:2405 ssl->f_recv(_timeout)() returned 1335 (-0xfffffac9)

I (13690) wifi: pm start, type:0

I (37586) mbedtls: ssl_tls.c:2404 in_left: 1340, nb_want: 2481

I (37586) mbedtls: ssl_tls.c:2405 ssl->f_recv(_timeout)() returned 1141 (-0xfffffb8b)

I (37590) mbedtls: ssl_tls.c:2417 <= fetch input

D (37609) mbedtls: ssl_tls.c:3103 handshake message: msglen = 2476, type = 11, hslen = 2476

I (37610) mbedtls: ssl_tls.c:3975 <= read record

D (37613) mbedtls: ssl_tls.c:4360 peer certificate #1:

D (37618) mbedtls: ssl_tls.c:4360 cert. version     : 3

...... all ok from now on but it took 20 seconds for the handshake!

projectgus, 2017-08-16:
Hi @lenzarda,

20 seconds is a long time for a handshake! To be clear, sometimes the handshake completes quickly, sometimes slowly, and sometimes not at all (how long do you wait for it?)

It looks like you have a timeout set on the mbedTLS connection (which is good), can you confirm? Are you able to post the exact code you're using somewhere?

Do you think the TCP connection may be quite lossy? Is there any possibility of getting a packet capture from either the WiFi router or the server side?

Angus

projectgus, 2017-08-31:
Hi @lenzarda,

I tried to reproduce this yesterday on the current IDF master branch by modifying the HTTPS example to point to the same server you are using, and to fail if a handshake takes more than 10 seconds. Over 1000 requests with no failures.

The code I used is here: https://gist.github.com/projectgus/4d6b8dd7b1249e01f96f9fd796543b46

There have been some improvements and fixes to mbedTLS since this was reported, so it's possible this issue has been resolved. Otherwise, is there anything you can provide to help us to reproduce? A wireshark/tcpdump packet capture of a failing session (from either the client or server end) would be very helpful.

Thanks,

Angus

lenzarda, 2017-09-05:
Thanks for your support! It works now. You can close this ticket.

closed 2017-09-05 by lenzarda