espressif/esp-idf #851: mcpwm_get_duty returns incomplete value

sunnyque, 2017-07-27:
having trouble to use MCPWM function mcpwm_get_duty

```c	//1. mcpwm gpio initialization
	mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, GPIO_PWM0A_OUT);

	//2. initial mcpwm configuration
	mcpwm_config_t pwm_config;
	pwm_config.frequency = 400; // frequency = 400Hz,
	pwm_config.cmpr_a = 0;		// duty cycle of PWMxA = 0
	pwm_config.cmpr_b = 0;		// duty cycle of PWMxb = 0
	pwm_config.counter_mode = MCPWM_UP_COUNTER;
	pwm_config.duty_mode = MCPWM_DUTY_MODE_0;
	mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config); // Configure PWM0A & PWM0B with above settings

	mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, 55.5f);
	mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_0); //call this each time, if operator was previously in low/high state

	printf("mcpwm check = %f\n", mcpwm_get_duty(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A));```

so mcpwm is set to 55.5 but get returns 55.000000 and this is wrong

also I've compared using logic analyzer 55.5 and 55.6 duty PWM output and it has difference for me when set, so problems seems just in get, value after point is lost somewhere

panfeng-espressif, 2018-02-22:
Hello sunnyque,
I am really quite sorry for the late response. 
I have tried your code and found that it really showed 55.00000 as the output duty. At the same time, I tried to use logic analyzer, the data read as 55.46%, quite close to the setting duty.
We have done some improvement after looking at your issue. But we just forget to reply to you. Hope the lastest IDF has solved your problem.
Thanks

sunnyque, 2018-02-22:
Thanks for your response, I have already switched to LED PWM that working well. Anyway it is good that issue has been fixed so closing issue

closed 2018-02-22 by sunnyque