espressif/esp-idf #852: Error in OpenOCD cfg: esp108 => esp32

MrBuddyCasino, 2017-07-28:
The file esp-idf/docs/api-guides/esp32.cfg contains this line:
`esp108 flashbootstrap 1.8`

This doesn't work anymore, OpenOCD will not start. It should say `esp32 flashbootstrap 1.8` instead.

krzychb, 2017-07-29:
Hi @MrBuddyCasino,

This doesn't work anymore because in last couple of days there were updates to OpenOCD. This ` esp-idf/docs/api-guides/esp32.cfg` file is now replaced with individual files per board / module type:

- [esp-wroom-32.cfg](https://github.com/espressif/openocd-esp32/blob/7a98088cfdbe60570c9866c7aefa65658ad3973a/tcl/board/esp-wroom-32.cfg)
- [esp32-wrover.cfg ](https://github.com/espressif/openocd-esp32/blob/7a98088cfdbe60570c9866c7aefa65658ad3973a/tcl/board/esp32-wrover.cfg)

For details please see revised and extended [JTAG Debugging](http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/index.html) documentation.

FayeY, 2017-08-01:
Hi @MrBuddyCasino , did the above suggestion help you resolve the issue?

MrBuddyCasino, 2017-08-01:
Looks like the documentation has been updated, so: yes. Thanks!

closed 2017-08-01 by MrBuddyCasino