espressif/esp-idf #853: Not able to connect to TI SensorTag to ESP32

patilsan90, 2017-07-28:
I am trying to connect to TI sensorTag cc2650 ( http://www.ti.com/tool/cc2650stk )
Its characteristics descriptor is given here
http://processors.wiki.ti.com/images/a/a8/BLE_SensorTag_GATT_Server.pdf

I was trying to read values from different sensors, but i am not able to. I am not able to read multiple characteristics from one service also.

If some demo example is given which will enable and read sensorTag value then it will be big help and I can proceed using ESP32

Thanks in advance

nkolban, 2017-07-28:
It isn't clear what application you have written to attempt to connect .... there are some sample ESP32 BLE app samples ... https://github.com/espressif/esp-idf/tree/master/examples/bluetooth

In addition, there are some C++ samples here ...

https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLE%20Tests

The ESP32 doesn't have anything "special" for the TI sensor tag ... however, if it is a BLE device, it should be able to be connected to as long as we write an application that connects with it.

patilsan90, 2017-07-28:
I have server device which has 3 characteristics in one service. I am able
to read the first characteristic of that service in client app. But how I
should read/write second n third characteristics. Please help me with demo
code flow to achieve the same.

https://github.com/espressif/esp-idf/tree/master/examples/bluetooth

https://github.com/nkolban/esp32-snippets/tree/master/
cpp_utils/tests/BLE%20Tests

nkolban, 2017-07-28:
I think I am hearing that you are looking for assistance writing an ESP32 application that achieves the goals that you have set for yourself.  This Github issues log is primarily for defects found in the ESP32 APIs as opposed to usage questions.   Where documentation on the usage of ESP32 is incomplete or wrong then that too is considered a defect.

The documentation on ESP32 BLE programming can be found here:

http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/index.html

Have you had a chance to study that in detail?  Are you able to comprehend the BLE sample applications supplied with the ESP-IDF?

If you think there is a code defect in the ESP-IDF then lets see if we can get to the bottom of that.  If our challenge is getting you up to speed on writing C or C++ applications that you are responsible for writing that use the exposed ESP-IDF APIs, then let me suggest that you create a forum thread on the ESP32 forum found here:

https://esp32.com/index.php

That is the place where the ESP32 community "hangs out" to offer assistance, guidance and generally chat about all things ESP32.

FHFS, 2017-07-28:
> I have server device which has 3 characteristics in one service. I am able to read the first characteristic of that service in client app. But how I should read/write second n third characteristics.
[esp32 forum post](https://esp32.com/viewtopic.php?f=2&t=2571)

closed 2017-08-01 by FayeY