espressif/esp-idf #854: No Internet access when in "APSTA" combined AP and station mode

valsinats42, 2017-07-28:
Hello, I'm trying to use the combination mode with ESP32 acting both as an access point and as a station.  Unfortunately, the Internet connection through lwip does not seem to be configured properly in this mode, so both DNS queries and connections to a fixed IP address fail. The connection works well when I configure the module to act as a station only.

To reproduce, I just changed line 90 in 'http_request_example_main.c" to use "WIFI_MODE_APSTA" instead of "WIFI_MODE_STA", and I get the attached log. Is there something I'm missing?

```
I (0) cpu_start: App cpu up.
I (1168) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1190) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1211) heap_alloc_caps: At 3FFB7CF0 len 00028310 (160 KiB): DRAM
I (1232) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1253) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1275) heap_alloc_caps: At 4009320C len 0000CDF4 (51 KiB): IRAM
I (1296) cpu_start: Pro cpu start user code
I (1353) cpu_start: Starting scheduler on PRO CPU.
I (202) cpu_start: Starting scheduler on APP CPU.
I (232) wifi: wifi firmware version: 407bb27
I (232) wifi: config NVS flash: enabled
I (232) wifi: config nano formating: disabled
I (232) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (242) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (272) wifi: Init dynamic tx buffer num: 32
I (272) wifi: Init dynamic rx buffer num: 32
I (272) wifi: wifi driver task: 3ffbda9c, prio:23, stack:4096
I (272) wifi: Init static rx buffer num: 10
I (272) wifi: Init dynamic rx buffer num: 32
I (282) wifi: Init rx ampdu len mblock:7
I (282) wifi: Init lldesc rx ampdu entry mblock:4
I (292) wifi: wifi power manager task: 0x3ffc2e64 prio: 21 stack: 2560
I (292) example: Setting WiFi configuration SSID bc...
I (302) wifi: wifi timer task: 3ffc3ee4, prio:22, stack:3584
I (372) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 0, 0
I (372) wifi: mode : sta (30:ae:a4:03:89:4c) + softAP (30:ae:a4:03:89:4d)
I (502) wifi: n:6 1, o:6 0, ap:6 1, sta:6 0, prof:6
I (1482) wifi: state: init -> auth (b0)
I (1482) wifi: state: auth -> assoc (0)
I (1492) wifi: state: assoc -> run (10)
I (1512) wifi: connected with bc, channel 6
I (3142) event: ip: 192.168.100.122, mask: 255.255.255.0, gw: 192.168.100.1
I (3142) example: Connected to AP
E (17042) example: DNS lookup failed err=202 res=0x0
I (18042) example: Connected to AP
E (31772) example: DNS lookup failed err=202 res=0x0
I (32772) example: Connected to AP
E (46522) example: DNS lookup failed err=202 res=0x0
I (47522) example: Connected to AP
```

FHFS, 2017-07-28:
did you configure for STA as well as AP?
esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
*note: never tried ap+sta, I could be totally off*

heyinling, 2017-07-30:
@valsinats42 which IDF reversion you're using? I tried on latest idf master and it works well:

```
I (302) example: Setting WiFi configuration SSID UTT-750...
I (302) wifi: wifi timer task: 3ffc3e94, prio:22, stack:3584
I (402) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 0, 0
I (412) wifi: mode : sta (24:0a:c4:00:ca:34) + softAP (24:0a:c4:00:ca:35)
I (3312) wifi: ap channel adjust o:1,1 n:10,2
I (3312) wifi: n:10 2, o:1 0, ap:10 2, sta:10 2, prof:1
I (4292) wifi: state: init -> auth (b0)
I (4532) wifi: state: auth -> assoc (0)
I (5032) wifi: state: assoc -> run (10)
I (5052) wifi: connected with UTT-750, channel 10
I (5712) event: ip: 192.168.17.104, mask: 255.255.255.0, gw: 192.168.17.1
I (5712) example: Connected to AP
I (5912) example: DNS lookup succeeded. IP=93.184.216.34
I (5912) example: ... allocated socket
```

The only thing need to pay attention is external AP should not allocate IP within the same subnet of SoftAP, which by default is `192.168.4.x`. but from log your STA IP is correct.

valsinats42, 2017-07-31:
Hi @heyinling,

We were using the latest v2.1 release of the IDF. I tested with a clean installation and it worked for me too. But then I found out we've added a "#define LWIP_NETIF_LOOPBACK 1" in "components/lwip/include/lwip/port/lwipopts.h", which seems to be breaking it. We're currently using the loopback functionality for our system tests. Is there a reason this option is not enabled by default or a reason it's breaking the normal routing? Thanks!

liuzfesp, 2017-08-13:
Hi @valsinats42, thanks for raising this issue, we will fix it

zhangyanjiaoesp, 2017-08-21:
@valsinats42  I have run the "http_request_example_main.c" to use "WIFI_MODE_APSTA", and added a "#define LWIP_NETIF_LOOPBACK 1", but I didn't reproduce your issue.
[image]

zhangyanjiaoesp, 2017-09-01:
Hi,@valsinats42, can you describe how you set the "#define LWIP_NETIF_LOOPBACK 1" in detail?

zhangyanjiaoesp, 2017-09-04:
Hi,@valsinats42, I have reproduce your problem, and I'm resolving it

valsinats42, 2017-09-04:
Hi, @zhangyanjiaoesp , thanks for that! Sorry for the lack of communication on my part, I'm currently on a tight schedule for other features of the product. We're currently using the workaround of disabling the loopback interface, but it would be nice if it would be soon supported. Thanks!

leonchang0000, 2017-09-04:
Hi.. Just want to ask.. Not sure if it's the same issue as you had here.
I'd like to setup ESP32 as APSTA mode to serve as an AP and STA concurrently.
So STA of my ESP32 will connect to a WIFI router (to have internet access) and another device (let's say device A) will connect soft-AP of the same ESP32.

My question is.. is it possible that this device A can have internet access.. 
From my tryout (including the latest esp-idf or the one from Jul 29).. both of them don't work...
(not even be able to ping WIFI router..)

Can anybody advise?

Thanks.

zhangyanjiaoesp, 2017-09-13:
Hi,@leonchang0000 , we have not support this function like you describe, because when in "APSTA" mode , ESP32 can't transmit data from "ap" netif to "sta"netif automatically.

closed 2017-11-21 by FayeY

auladinamica, 2017-11-22:
Hi guys : I have this scenario : I have one esp32 called (server) that is conected to my wifi like a sation mode, and i have 4 esp32 that i need send data to the server, how is the best way to connect this 4 esp to the server ? I think create in the server like AP and the 4 ESP (clients) connect  to this server... Where can i get the code to do this ? Is there a sample o somethig like that ?

Thanks In advance

github-sls, 2019-04-17:
> 
> 
> Hi,@valsinats42, I have reproduce your problem, and I'm resolving it

May i know the commit where it has been fixed ? Was it the exact problem told by valsinats42 ?

hiproz, 2021-06-04:
i use ESP-IDF v4.4-dev-1404-gc13afea63-dirty  for esp32c3， i had the same issue that cant visit the internet in apsta mode.  @zhangyanjiaoesp so, is it not been supported still:

> in "APSTA" mode , ESP32 can't transmit data from "ap" netif to "sta"netif automatically.