espressif/esp-idf #857: Regression: BLE Server now no longer showing exposed services

nkolban, 2017-07-31:
I have a library written in C++ that prior to the latest refresh of the ESP-IDF allowed me to become a BLE server.  See ... 

https://www.youtube.com/watch?v=2_vlF_02VXk

When I try this same app using the build as of the date of this posting, I find that the server still appears to be found by a client, but I now no-longer see the service and (of course) none of its characteristics.  I have added the NVS initialization but that hasn't helped.

There are some errors being logged to the output such as:

* `E (2356) BT: osi_alarm_set chg period error`
* `E (2462) BT: osi_alarm_cancel failed to stop timer`

but I don't know if those are innocuous or not ...

```
make monitor
MONITOR
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:3584
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:11968
entry 0x40078cd8
I (1570) heap_init: Initializing. RAM available for dynamic allocation:
I (1571) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (1573) heap_init: New heap initialised at 0x3ffaff10
I (1618) heap_init: At 3FFCB1F8 len 00014E08 (83 KiB): DRAM
D (1667) heap_init: New heap initialised at 0x3ffcb1f8
I (1711) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1762) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1813) heap_init: At 4009333C len 0000CCC4 (51 KiB): IRAM
D (1862) heap_init: New heap initialised at 0x4009333c
D (2032) heap_init: New heap initialised at 0x3ffe0440
D (2032) heap_init: New heap initialised at 0x3ffe4350
D (2032) Task: >> runTask
D (2032) SampleServer: Starting BLE work!
D (2037) nvs: nvs_flash_init_custom start=9 count=6
D (2206) nvs: nvs_open phy 0
D (2206) nvs: nvs_get cal_version 4
D (2206) nvs: nvs_get_str_or_blob cal_mac
D (2207) nvs: nvs_get_str_or_blob cal_data
D (2209) nvs: nvs_close 1
D (2228) nvs: nvs_open bt_config.conf 1
D (2229) nvs: nvs_get_str_or_blob bt_cfg_key
D (2229) nvs: nvs_close 2
D (2229) nvs: nvs_open bt_config.conf 1
D (2231) nvs: nvs_set_blob bt_cfg_key 216
D (2236) nvs: nvs_close 3
D (2258) nvs: nvs_open bt_config.conf 1
D (2259) nvs: nvs_set_blob bt_cfg_key 216
D (2260) nvs: nvs_close 4
D (2260) nvs: nvs_open bt_config.conf 1
D (2261) nvs: nvs_set_blob bt_cfg_key 216
D (2265) nvs: nvs_close 5
D (2266) nvs: nvs_open bt_config.conf 1
D (2270) nvs: nvs_set_blob bt_cfg_key 216
D (2275) nvs: nvs_close 6
D (2277) BLEServer: >> registerApp - 0
D (2283) BLE: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_REG_EVT
D (2286) BLEUtils: GATT ServerEvent: ESP_GATTS_REG_EVT
D (2291) BLEUtils: [status: ESP_GATT_OK, app_id: 0]
D (2295) BLEServer: >> handleGATTServerEvent: ESP_GATTS_REG_EVT
D (2301) BLEServer: << handleGATTServerEvent
D (2305) BLEServer: << registerApp
D (2308) BLEServer: >> createService - 00001234-0000-1000-8000-00805f9b34fb
D (2315) BLEService: >> executeCreate() - Creating service (esp_ble_gatts_create_service) service uuid: 00001234-0000-1000-8000-00805f9b34fb
D (2328) BLE: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CREATE_EVT
D (2334) BLEUtils: GATT ServerEvent: ESP_GATTS_CREATE_EVT
D (2339) BLEUtils: [status: ESP_GATT_OK, service_handle: 0x28, service_id: [uuid: 00001234-0000-1000-8000-00805f9b34fb, inst_id: 0]]
D (2351) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
E (2356) BT: osi_alarm_set chg period error

D (2362) BLEService: >> setHandle - Handle=0x28, service UUID=00001234-0000-1000-8000-00805f9b34fb)
D (2370) BLEService: << setHandle
D (2373) BLEServer: << handleGATTServerEvent
D (2377) BLEService: << executeCreate
D (2380) BLEServer: << createService
D (2384) BLEService: >> addCharacteristic()
D (2387) BLEService: Adding characteristic (esp_ble_gatts_add_char): uuid=000099aa-0000-1000-8000-00805f9b34fb to service: UUID: 00001234-0000-1000-8000-00805f9b34fb, handle: 0x28
D (2403) BLEService: << addCharacteristic()
D (2407) BLECharacteristic: >> setValue: length=12, data=48656c6c6f20576f726c6421, characteristic UUID=000099aa-0000-1000-8000-00805f9b34fb
D (2419) BLECharacteristic: << setValue
D (2423) BLECharacteristic: >> addDescriptor(): Adding UUID: 00002902-0000-1000-8000-00805f9b34fb, handle: 0xffff to UUID: 000099aa-0000-1000-8000-00805f9b34fb, handle: 0xffff Read Write Broadcast Notify Indicate 
D (2442) BLECharacteristic: << addDescriptor()
D (2446) BLEService: >> start(): Starting service (esp_ble_gatts_start_service): UUID: 00001234-0000-1000-8000-00805f9b34fb, handle: 0x28
D (2458) BLECharacteristic: >> executeCreate()
E (2462) BT: osi_alarm_cancel failed to stop timer

D (2467) BLECharacteristic: Registering characteristic (esp_ble_gatts_add_char): uuid: 000099aa-0000-1000-8000-00805f9b34fb, service: UUID: 00001234-0000-1000-8000-00805f9b34fb, handle: 0x28
D (2484) BLE: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_ADD_CHAR_EVT
D (2490) BLEUtils: GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
D (2496) BLEUtils: [status: ESP_GATT_OK, attr_handle: 0x2a, service_handle: 0x28, char_uuid: 000099aa-0000-1000-8000-00805f9b34fb]
D (2507) BLEServer: >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
D (2513) BLECharacteristic: >> setHandle: handle=0x2a, characteristic uuid=000099aa-0000-1000-8000-00805f9b34fb
D (2523) BLECharacteristic: << setHandle
D (2527) BLEServer: << handleGATTServerEvent
D (2531) BLEDescriptor: >> executeCreate(): UUID: 00002902-0000-1000-8000-00805f9b34fb, handle: 0xffff
D (2540) BLE: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_ADD_CHAR_DESCR_EVT
D (2547) BLEUtils: GATT ServerEvent: ESP_GATTS_ADD_CHAR_DESCR_EVT
D (2553) BLEUtils: [status: ESP_GATT_OK, attr_handle: 0x2b, service_handle: 0x28, char_uuid: 00002902-0000-1000-8000-00805f9b34fb]
D (2564) BLEServer: >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_DESCR_EVT
E (2567) BT: osi_alarm_set chg period error

D (2576) BLEDescriptor: >> setHandle(0x2b): Setting descriptor handle to be 0x2b
D (2582) BLEDescriptor: << setHandle()
D (2586) BLEServer: << handleGATTServerEvent
D (2590) nvs: nvs_open bt_config.conf 1
D (2594) nvs: nvs_set_blob bt_cfg_key 216
D (2599) nvs: nvs_close 7
D (2599) BLEUtils: Received a GAP event: ESP_GAP_BLE_LOCAL_ER_EVT
D (2605) BLEServer: BLEServer ... handling GAP event!
D (2610) BLEDescriptor: << executeCreate
D (2614) BLECharacteristic: << executeCreate
D (2619) BLE: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_START_EVT
D (2624) BLEUtils: GATT ServerEvent: ESP_GATTS_START_EVT
D (2629) BLEUtils: [status: ESP_GATT_OK, service_handle: 0x28]
D (2635) BLEServer: >> handleGATTServerEvent: ESP_GATTS_START_EVT
D (2641) BLEServer: << handleGATTServerEvent
D (2645) BLEService: << start()
D (2648) BLEAdvertising: >> setServiceUUID - 00001234-0000-1000-8000-00805f9b34fb
D (2655) BLEAdvertising: << setServiceUUID
D (2659) BLEAdvertising: >> start
D (2662) BLEAdvertising:  - Service: service_uuid_len=16, p_service_uuid=0x3ffdd82a (data=fb349b5f800000800010000034120000)
D (2674) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
E (2675) BT: osi_alarm_cancel failed to stop timer

D (2684) BLEUtils: [status: 0]
D (2687) BLEServer: BLEServer ... handling GAP event!
D (2693) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (2699) BLEUtils: [status: 0]
D (2701) BLEServer: BLEServer ... handling GAP event!
D (2706) BLEAdvertising: << start
D (2709) SampleServer: Advertising started!
D (46254) BLE: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CONNECT_EVT
D (46255) BLEUtils: GATT ServerEvent: ESP_GATTS_CONNECT_EVT
D (46256) BLEUtils: [conn_id: 0, remote_bda: 63:ec:f0:00:b7:47, is_connected: 1]
D (46263) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
D (46269) BLEServer: << handleGATTServerEvent
```

nkolban, 2017-07-31:
As a test, I cloned the pre-release of 2.1 release found here:

https://github.com/espressif/esp-idf/commit/3bd3640d60a0aff719470dd870d8ef3527ee9953

and recompiled everything ... and all that worked previously now works again.  This seems to be more evidence that something has regressed.

nkolban, 2017-07-31:
I performed another test ... this time I download release 2.1 as opposed to master ...

https://github.com/espressif/esp-idf/commit/27574a31e7c41f6e22e10e9b9d7713c6d49f795d

And THIS built works just fine.  This seems to show me that something between 2.1 and current master has regressed.

TianHao-Espressif, 2017-08-17:
@nkolban , please check your CONFIG_TIMER_QUEUE_LENGTH in sdkconfig.h. Is there differences between your two version?
 If CONFIG_TIMER_QUEUE_LENGTH (default is 10) is too little, it may cause the alarm reference called failed.

nkolban, 2017-08-17:
Howdy my friend,
I just tested with the very latest build and the problem remains.  The issue isn't the generation of the alarm references ... the issue that when I run my BLE Server, I can find the server because it is advertising.  I can then connect to the server ... however, the only services that are shown when connected are:

0x1801 - Generic Attribute
0x1800 - Generic Access

My own customer service does not show up.   If I clean the whole environment and the rebuild everything but this time using ESP-IDF release 2.1 (and not the ESP-IDF master release which is what I had been using), the very same applications runs but this time I see my custom exposed service.

So ... compiling an app with ESP-IDF (master) ... code runs, I do NOT see my exposed service.
Re compiling the same app with ESP-IDF (2.1) ... code runs, I DO see my exposed service.

I am using Android and nRF Connect tool for testing.

TianHao-Espressif, 2017-08-17:
@nkolban, thanks for your description. In master branch, we will add some new feature and will modify much codes for V3.0. So there's some modification in master branch(I think your problem due to the service cache). Please use the branch release/v2.1 as your common branch. Besides, we will look for the root cause of this problem and fix it.

coder543, 2017-10-03:
@TianHao-Espressif I'm experiencing a [very similar issue here.](https://github.com/nkolban/esp32-snippets/issues/109) Solving this issue might solve my issue as well. Any help is appreciated.

blueMoods, 2017-10-19:
@nkolban  Hello, because we can not connect to youtube, please put your code to github. Thanks.

nkolban, 2017-10-19:
Howdy ... all code is available here

https://github.com/nkolban/esp32-snippets

blueMoods, 2017-10-19:
@nkolban  I add Makefile at esp32-snippets/cpp_utils/tests/BLETests/
The contents of the Makefile are as follows:
PROJECT_NAME := BLETests

COMPONENT_ADD_INCLUDEDIRS := components/include

include $(IDF_PATH)/make/project.mk

But I make failed ,the log are as follows:
/home/baohongde/github/customer/esp32-snippets/cpp_utils/tests/BLETests/build/esp32/libesp32.a(cpu_start.o):(.literal.main_task+0xc): undefined reference to `app_main'
/home/baohongde/github/customer/esp32-snippets/cpp_utils/tests/BLETests/build/esp32/libesp32.a(cpu_start.o): In function `main_task':
/home/baohongde/develop/esp-idf/components/esp32/./cpu_start.c:392: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
/home/baohongde/develop/esp-idf/make/project.mk:359: recipe for target '/home/baohongde/github/customer/esp32-snippets/cpp_utils/tests/BLETests/build/BLETests.elf' failed
make: *** [/home/baohongde/github/customer/esp32-snippets/cpp_utils/tests/BLETests/build/BLETests.elf] Error 1

Could you please help to find where i was wrong. And how to write Makefile.

coder543, 2017-10-20:
It sounds like your Makefile is not actually compiling and linking the `esp32-snippets/cpp_utils/tests/BLETests/main.cpp` into your executable, which has the `app_main` function in it.

nkolban, 2017-11-19:
@blueMoods For issues on the esp32-snippets projects, please create issues here:

https://github.com/nkolban/esp32-snippets/issues

nkolban, 2017-12-12:
A number of us working in BLE land continue to experience the problem and are able to recreate at will.  We have found that when we compile with:

```
CONFIG_OPTIMIZATION_LEVEL_DEBUG=
CONFIG_OPTIMIZATION_LEVEL_RELEASE=y
```

works but compiling the exact same code:

```
CONFIG_OPTIMIZATION_LEVEL_DEBUG=y
CONFIG_OPTIMIZATION_LEVEL_RELEASE=
```

fails.  Our recreate recipe is to use the sample called "SampleWrite.cpp" found here:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleWrite.cpp

I will be delighted to assist anyone in development at any time to get a recreate going.

chegewara, 2018-02-21:
Hi, two months later this issue is still not solved. Thanks

blueMoodBHD, 2018-03-21:
@nkolban @chegewara Does the problem still exist?

chegewara, 2018-03-21:
@blueMoodBHD Hi, yes the issue is still not being solved and the only way is workaround from this post
https://github.com/espressif/esp-idf/issues/857#issuecomment-350953996

blueMoodBHD, 2018-03-23:
@nkolban @chegewara I find a bug in esp32-snippets.

At cpp_utils/BLEService.cpp line 75, you didn't set `srvc_id.is_primary`. So the services you created may be Second Service and will not show.

In release mode, `srvc_id.is_primary`  will be initialed to a rand value, mostly not 0, so it works well. But in debug mode, it will be initialed to 0, causing the service not showing.

Please add this line and test again, thanks!
`srvc_id.is_primary = true;`

chegewara, 2018-03-26:
@blueMoodBHD Seems it fixed the issue, i owe you beer. Many thanks.

The problem was it sometimes worked and sometimes it did not, depend on example, and to be honest i did not know what is difference between primary and secondary service (still dont know).

blueMoodBHD, 2018-03-27:
@chegewara There are two types of services: primary and secondary. A primary service is a service that provides the primary functionality of a device. A secondary service is a service that provides auxiliary functionality of a device and is referenced from at least one primary service on the device.

So, if a secondary service is referenced from no primary service, it will not show.

chegewara, 2018-03-27:
@blueMoodBHD Thanks a lot for info and help with our bug.

closed 2018-04-04 by FayeY