espressif/esp-idf #859: ESP is extremely sensitive to AC Loads

Raienryu97, 2017-07-31:
Hello
A few days ago, I had opened up a [forum post](https://esp32.com/viewtopic.php?f=19&t=2541) on the ESP 32 forum regarding an issue I had with Interrupts being triggered when AC Load was toggled nearby. 
The code I have used to test this is in the first post of the thread.  I have tried the suggestions mentioned there, and still had no success, still having random interrupts being triggered. 

I have tried many debouncing methods I both software and hardware , all failed.

I have also tested using ESP-IDF code (instead of arduino as on the forum) and the issue still persisted. (Will post code if required, it is quite a simple logic, so didn't post it here)

I have tried all the various power supplies too , a dedicated adapter, laptop's USB port , issue remains, so I assume it's not a power line problem.  

There was also [another reply today](https://esp32.com/viewtopic.php?f=19&t=2541#p12219) to the thread with another person experiencing similar (and worse) issues as mine which leads me to think that there  probably could be  an issue fundamentally with the esp32 itself.

Any suggestions or comments on this issue would be great. 

Thanks.

Spritetm, 2017-07-31:
Please post your hardware setup and code, please, so we can see if we can replicate it.

Raienryu97, 2017-07-31:
Hey @Spritetm thanks for the response.  Here are the details.

**Hardware Setup**
==================
* I use this [ESP 32 Dev Board](https://www.aliexpress.com/store/product/Goouuu-esp32-development-board-wireless-WiFi-Bluetooth-2-and-1-dual-core-CPU-core-board-ESP/2340056_32799809384.html) 
* I have in use two GPIOs, 
  * GPIO34 for the input from a button which is hardware debounced by [this circuit](https://hackadaycom.files.wordpress.com/2015/11/debounce-sch.png)
  * GPIO17 which is an output pin connected to a 2 channel Relay module (of which I use only 1)
* The relay output is connected to a standard 15W 230V lamp
* Currently using a very good quality USB Cable as a power source for my ESP32 as it makes my debugging easier 

**Code**
=============
```C
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"

#define LED_GPIO 17
#define GPIO_INPUT_IO_1     34 //Button
#define GPIO_INPUT_PIN_SEL  ((uint64_t)(((uint64_t)1)<<34))
#define ESP_INTR_FLAG_DEFAULT 0

volatile bool shouldPrint = 0;
volatile int state = 0;
volatile unsigned long int current,last = 0;

static void IRAM_ATTR gpio_isr_handler(void* arg)
{
  //we wait for debounce
  current = xTaskGetTickCountFromISR();
  if(abs(current - last) > 20){
    shouldPrint = 1;
    state = !state;
  }
  last = current;
}

void app_main()
{
  gpio_config_t io_conf;

  //interrupt of rising edge
  io_conf.intr_type = GPIO_PIN_INTR_POSEDGE;
  //bit mask of the pins, use GPIO4/5 here
  io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
  //set as input mode
  io_conf.mode = GPIO_MODE_INPUT;
  gpio_config(&io_conf);

  //install gpio isr service
  gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
  //hook isr handler for specific gpio pin
  gpio_isr_handler_add(GPIO_INPUT_IO_1, gpio_isr_handler, (void*) GPIO_INPUT_IO_1);

  gpio_pad_select_gpio(LED_GPIO);
  /* Set the GPIO as a push/pull output */
  gpio_set_direction(LED_GPIO, GPIO_MODE_OUTPUT);

  while(1){
    if(shouldPrint){
      gpio_set_level(LED_GPIO,state);
      printf("Set LED state to: %d\n",state);
      shouldPrint = 0;
    }
  }

}
```
Issue Reproduction Steps
=====================
* Simply plug and unplug the bulb adapter into/from the socket
* Plug something else into a neighbouring socket
When either of the above is done, an Interrupt is triggered and the Bulb toggles/message is printed on the monitor

Hope this makes it easier for you to replicate the issue

negativekelvin, 2017-07-31:
That isn't really software debouncing, that just limits how close two interrupts can occur.  You are using an opto isolated relay, so that's good.  What are you using to power the relay board?

Raienryu97, 2017-07-31:
@negativekelvin  I have connected the V5 pin of the ESP32 to the Vcc of the Relay

negativekelvin, 2017-07-31:
Software debounce usually samples the input multiple times to filter out short spikes/noise. You should try to disconnect one wire at a time and see which eliminates the problem.

Raienryu97, 2017-07-31:
I have absolutely no problem when an AC Load is not connected ( i.e when the bulb is not placed in it's holder) . I have tried various AC Loads. All of them have the issue.

krzychb, 2017-07-31:
>  I have connected the V5 pin of the ESP32 to the Vcc of the Relay, 

I would provide a separate power supply to the relay board.

bvernoux, 2017-07-31:
It is forbidden to connect a GPIO directly to a relay you can kill the MCU
with such wrong usage !!
See a good schematic from Olimex ESP32-EVB to do that correctly:
https://github.com/OLIMEX/ESP32-EVB/blob/master/HARDWARE/REV-B/ESP32-EVB_Rev_B.pdf
See the part Relays in the PDF schematic you shall do that mainly using a
transistor to drive the relay and the relay shall have some diode to avoid
feedback ...

negativekelvin, 2017-07-31:
@bvernoux relax that relay board has optos on it

@Raienryu97  you should still see if you can prevent it with the ac load present by disconnecting a single jumper

Raienryu97, 2017-07-31:
@negativekelvin @krzychb thanks for the suggestions.  I will try that and let you guys know. 

I would also like to know ( maybe to use as a workaround) if it is possible to control the sensitivity at which a pin could be triggered. If the noise which causes the interrupt has a lower value than what would occur from a button , I guess I could circumvent the issue.

copercini, 2017-07-31:
One more guess: Did you tried another pin instead of GPIO34? another one out of 34-39 range?

Raienryu97, 2017-08-02:
Hello.
Thank you very much for the advice. I have tested the suggestions provided.

* I have tried pins outside of 34-39 (specifically 16 and 17) and the **issue persists**
* I have tried removing various jumpers trying to isolate the issue and what I have observed from this is that there is no issue when the AC load is not connected to power (the issue doesn't occur when bulb adapter is not plugged in), but obviously that defeats the purpose of the application. 
* I have also tried using a separate power source for the relay board and have made an observation 
  regarding this. **Issue persists but is a bit lesser**

   * The observation  is that when the code is running on Arduino Uno board and relay is powered by ESP32 (esp32 is ONLY used to power relay) , then there is absolutely no issue
BUT when the code is running on ESP32 and relay is powered by Arduino Uno (Uno ONLY for powering relay), the issue persists.

feredy2011, 2017-08-22:
are you using a pullup resistor (4.7k) for button ? perhaps this issue is related to this !

Raienryu97, 2017-09-11:
@feredy2011 , I am using [this circuit](https://hackadaycom.files.wordpress.com/2015/11/debounce-sch.png) for the button. Sorry for the late response. But I had gotten a solution and have forgotten to close the comment in time. Thanks to everyone for their inputs. They helped in further decreasing the issue.

#### Note to the people in the future who have been troubled with this issue just like me
My issue was reduced A LOT in fact almost completely nullified simply by adding an RC series (snubber) to the contacts of the relay. Note that this will cause a bit of leakage current to be flowing even when the relay switch is turned off.

closed 2017-09-11 by Raienryu97

benjaminaigner, 2017-09-11:
@Raienryu97 
If you would like to get rid of this snubber circuit, maybe have a look at bistable relays (e.g. ADW1103), which are only driven for a few milliseconds to change the relay contact position.

Of course, this problem might persist. BUT: it is mainly caused by the PCB design.
I have currently a board with 6 bistable and 2 "normal" relays, no problems at all, even if you switch on all relays at one.

Raienryu97, 2017-09-11:
@benjaminaigner, definetely will take a look. Getting rid of the snubber will prevent the leakage. Thanks for the suggestion.

eumb, 2018-12-04:
@Raienryu97  I am having similar issues as you described when using a solid state relay to drive a motor. Whenever the motor stops some inputs will get triggered. 
My question is: how have you connected your snubber? Is it on the relay contacts in parallel or on the load ( e.g motor)?

a-c-sreedhar-reddy, 2019-01-08:
@edmund-huber did you find the solution

tichaonax, 2020-08-15:
I was having similar issue where the blinker set for half a second interval would slow down noticeably eventually shutting down the webserver not responsive. The problem dissappeared after after l changed to a different USB power supply. I came to notice that the problem only occurred when l remove the esp32 from my development lapton USB to a standalone power supply.