espressif/esp-idf #860: `uart_read_bytes()` does not flush hardware FIFO

psiphi75, 2017-07-31:
It seems that when calling `uart_read_bytes()` the hardware FIFO buffer is not flushed, [see this thread](https://esp32.com/viewtopic.php?f=2&t=2594).  This causes problems when you need to quickly react to events, such as the `UART_PATTERN_DET` event.

It causes the following issue: I have some time critical code where I need to read data from the UART, wait for a specific character (using `UART_PATTERN_DET`), read the buffer up until the detected character and then, if conditions are met, write a new set of characters to UART TX (within 4 milliseconds). The only issue is that when an `UART_PATTERN_DET` event is triggered and I use `uart_read_bytes()` no characters are in the buffer. I set `ticks_to_wait` to zero, because this data should be in the UART ringbuffer. I would expect that the `uart_read_bytes()` to return all the characters up until the `UART_PATTERN_DET` event was triggered.

FayeY, 2017-11-28:
Hi @psiphi75  , this issue should have been fixed in the latest IDF. If it is not, please feel free to reopen this issue. Thanks.

psiphi75, 2017-12-06:
Hi,

I have tried this with version 2.1.1 of the ESP IDF and I still get the same issue.  

Below is some code to reproduce the issue.  You just need to connect GPIO 4 and 15 together.  The sequence `0, 1, 2, ... 254, 255, 0, 1, ...` is constantly written to the UART.  The `uart_enable_pattern_det_intr` listens for one `0` character, then reads the buffer.  I expect the last character in the buffer to be `0`, or a higher number, but not `255` or less.  

However, everyone once in a while a `255`, or `254` character is printed, this shows that once the `UART_PATTERN_DET` event is triggered, then `uart_read_bytes()` doesn't flush all the bytes from the UART read buffer, likely from the hardware buffer. 

```C
#include "driver/uart.h"
#include "esp_log.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/queue.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "soc/uart_struct.h"
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

static const char *TAG = "uart_events";

static QueueHandle_t uart1_queue;

static void uart_event_task(void *pvParameters) {
  uart_event_t event;
  uint8_t *data = (uint8_t *)malloc(1024);
  uint8_t i = 0;
  for (;;) {
    data[0] = i++;
    uart_write_bytes(UART_NUM_1, (const char *)data, 1);
    if (xQueueReceive(uart1_queue, (void *)&event, (portTickType)1)) {
      if (event.type == UART_PATTERN_DET) {
        int len = uart_read_bytes(UART_NUM_1, data, 1024, 1);
        if (len > 0) {
          ESP_LOGI(TAG, "last byte read : %d", (int)data[len - 1]);
        }
      }
    }
  }
}

void app_main() {
  uart_config_t uart_config = {
      .baud_rate = 19200,
      .data_bits = UART_DATA_8_BITS,
      .parity = UART_PARITY_DISABLE,
      .stop_bits = UART_STOP_BITS_1,
      .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
      .rx_flow_ctrl_thresh = 122,
  };
  uart_param_config(UART_NUM_1, &uart_config);
  uart_driver_install(UART_NUM_1, 2048, 2048, 10, &uart1_queue, 0);
  uart_set_pin(UART_NUM_1, 4, 15, 18, 19);
  uart_enable_pattern_det_intr(UART_NUM_1, 0, 1, 10000, 10, 10);
  xTaskCreate(uart_event_task, "uart_event_task", 2048, NULL, 12, NULL);
}

```

psiphi75, 2017-12-11:
@FayeY Can you please re-open this issue, in my view it's not solved.  Thanks.

psiphi75, 2017-12-17:
@FayeY do you have the specific commit where this is fixed, or is not public yet?  I would be glad to test the commit.

bcanustundag, 2018-01-02:
I'm actually facing the same problem. Before entering deepsleep, I flush UART & delete driver using API functions. After esp32 wakes up and I initialize same UART driver and flush rx buffer using uart_flush. However even the driver is deleted and buffer is (soft) flushed, rx buffer is full of values from the last use.

I am certain that hardware fifo buffers cannot be flushed with a uart_flush() call.

igrr, 2018-01-08:
@psiphi75 The fix has been merged in 4052803e161ba06d1cae8d36bc66dde15b3fc8c7.

jack0c, 2018-03-21:
@FayeY Try to close this.

closed 2018-03-21 by FayeY

psiphi75, 2018-05-18:
I have tested this an it appears to work okay.  Thanks for the bug fix.