espressif/esp-idf #861: 关于WIFI_PROTOCOL_LR

lixx5427, 2017-07-31:
在ESP32的文档中说存在一个长距离（Long Range）工作模式。我在程序中试了试，看起来可以启动这个模式，也可以发出包来，但是其他WiFi设备好像接收不到发出来的数据包。我想知道这个模式的一点细节，即如何实现了长距离？是否是通过发送端降低DAC采样速率做到的？

liuzfesp, 2017-08-13:
HI @lixx5427 Long range模式只有在连接两端均为esp32时才能够使用，与非esp32设备连无法通讯。

lixx5427, 2017-08-13:
Hi @liuzfesp 非常感谢，但我需要让Long Range的ESP32和标准WiFi设备通信。我认为无论ESP32的Long Range采用了WiFi的哪些特性，我相信底层的WiFi调制、信道编码都应该是和标准WiFi兼容的。所以Long Range ESP32和标准WiFi应该能够通信，只是标准WiFi也应该启动同样的特性。

不知道我理解的对不对？如果正确，您能告诉我这个特性是什么？是不是WiFi标准上所说的DAC降采样？希望您能不吝赐教，非常感激。

hackrid, 2017-09-15:
There is definitely some info missing regarding the mysterious long range mode ☺️

Maybe it offers even more potential when it is discovered that various aspects of the WiFi/bt/LR radio can be tuned (modulation, spreading, frequency, power)😇 
would maybe offer a new level of universality

hackrid, 2017-09-15:
Maybe nrf24l01 compatibility is not really far from reach from btle

hackrid, 2017-09-24:
John Lee just posted some details about the LR mode [here](https://www.esp32.com/viewtopic.php?f=2&t=3096):

> sensitivity of -105 dBm with 125 kbps rate

closed 2017-11-21 by FayeY