espressif/esp-idf #866: ESP-IDF 2.1: Missing new BLE features

frax84, 2017-08-01:
Hello,
many of the new BLE feeatures of the release 2.1 of IDF are missing.
In particular:
- No possibility to enable just BLE (only BTDM)
- No APIs for set tx power

This is probably due (at least in part) to the old version of bt.h library available in 2.1 release. The version in the master branch has the missing APIs for BLE power settings.

Regards,
frax

TianHao-Espressif, 2017-08-02:
@frax84 , for your questions:

> No possibility to enable just BLE (only BTDM)

When we finish test some other things, we will open "enable ble/classic bt only". But there's no obviously difference between BTDM and BLE(or Classic BT).

>No APIs for set tx power

There's  BLE TX power APIs in master branch.

frax84, 2017-08-02:
@TianHao-Espressif 

From the 2.1 release changelog:

> Memory optimization
>BT and BLE can be enabled individually

If tests are still unfinished it's no problem, but if the changelog says that they can be enabled individually you should remove it. People have to trust changelog, otherwise it is useless :)
Moreover, even if there is no change in terms of functionality between BTDM and BLE there should be some difference in terms of both memory and power consumption, shouldn't be?

About BLE TX power APIs it's just my bad. For some reason i believed they have to be in release 2.1 but it was totally  nonsense supposition.

frax84, 2017-08-02:
@TianHao-Espressif 
i tried to update the esp-idf to master branch to use the function esp_ble_tx_power_set but i have an error during compiling.

>C:/ESP32IDF/esp-idf-v2.1/components/bt/bt.c:426: undefined reference to `ble_txpwr_set'

I tried to check where the function is defined but the only reference to the function is in bt.c:

`extern int ble_txpwr_set(int power_type, int power_level);`

There is no code for this function anywhere in esp-idf. Where i'm wrong?

igrr, 2017-08-02:
Have you done `git submodule update` after pulling the latest version? What is the output of `git status` and `git submodule` in `$IDF_PATH`?

frax84, 2017-08-02:
Ok,
some news about it. I was working in 2.1 branch where i updated the only BT component. Probably i made a fuss, breaking some reference.
I re-downloaded the esp-idf (master branch) and verified that everything was fine with `git status` and `git submodule`.
I tried to `make` the project but i encountered the same problem of this post:
[https://github.com/espressif/esp-idf/issues/828](url)
Using the workaround of the same post i had been able to succesfully compile.

NOTE:
For people like me that need ADC2 and used the appended APIs like indicated in this post:
[https://github.com/espressif/esp-idf/issues/461](url)
Attention needs to be paid because the main branch of esp-idf has a new rtc_module.c. Procedure needs to be repeated on this new file (you cannot copy the old modified one).

closed 2017-08-02 by frax84