espressif/esp-idf #868: esp_wifi_start + SDMMC usage cause brownout and rebooting errors

Staubgeborener, 2017-08-01:
So i tested the sd card example in 1-bit mode. Everything works nice. So i thought, i could implement the usage in my own code. Everyone who wants to see it: [Link](https://gist.github.com/Staubgeborener/83f06f83ef43e38112b0b6d8d50db3a8) (it's a mess, i know. but it worked)

So it works the whole time. Also with SD Card Usage with SPI. I copied the necessary codelines from the example in my code (initializing sd card, mounting, save data, close file, unmount). And this happens:
```C++
Name: SS08G
Type: SDHC/SDXC
Speed: default speed
Size: 7583MB
CSD: ver=1, sector_size=512, capacity=15529984 read_bl_len=9
SCR: sd_spec=2, bus_width=5
I (2755) wifi: wifi firmware version: 0e33bec
I (2755) wifi: config NVS flash: enabled
I (2755) wifi: config nano formating: disabled
I (2765) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2775) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2795) wifi: Init dynamic tx buffer num: 32
I (2795) wifi: Init data frame dynamic rx buffer num: 32
I (2795) wifi: Init management frame dynamic rx buffer num: 32
I (2805) wifi: wifi driver task: 3ffd9870, prio:23, stack:4096
I (2805) wifi: Init static rx buffer num: 10
I (2815) wifi: Init dynamic rx buffer num: 32
I (2815) wifi: Init rx ampdu len mblock:7
I (2815) wifi: Init lldesc rx ampdu entry mblock:4
I (2825) wifi: wifi power manager task: 0x3ffdeb4c prio: 21 stack: 2560
E (2825) wifi: wifi_set_chan_range 1733 ret=1
I (2835) wifi: wifi timer task: 3ffdfba4, prio:22, stack:3584
W (2835) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration

Brownout detector was triggered
```
-> see line 227 - 303 in code
Initializing sd card works. But soon as the esp should start wifi
``ESP_ERROR_CHECK(esp_wifi_start()); ``
the brownout appears.

So i tried 
``ESP_ERROR_CHECK(esp_wifi_start()); ``
in the sd card example: [Link](https://gist.github.com/Staubgeborener/18ebb80af40eb9b448269dbdb4af954d)
And here i've got a reboot loop. 

```C++
I (45) boot: ESP-IDF v3.0-dev-205-g9314bf0d-dirty 2nd stage bootloader
I (45) boot: compile time 20:29:31
I (49) boot: Enabling RNG early entropy source...
I (65) boot: SPI Speed      : 40MHz
I (78) boot: SPI Mode       : DIO
I (90) boot: SPI Flash Size : 4MB
I (103) boot: Partition Table:
I (114) boot: ## Label            Usage          Type ST Offset   Length
I (137) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (160) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (183) boot:  2 factory          factory app      00 00 00010000 00100000
I (207) boot: End of partition table
I (220) boot: Disabling RNG early entropy source...
I (237) boot: Loading app partition at offset 00010000
I (255) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0e474 ( 58484) map
I (345) esp_image: segment 1: paddr=0x0001e49c vaddr=0x3ffb0000 size=0x01b74 (  7028) load
I (354) esp_image: segment 2: paddr=0x00020018 vaddr=0x400d0018 size=0x3e554 (255316) map
0x400d0018: _flash_cache_start at ??:?

I (637) esp_image: segment 3: paddr=0x0005e574 vaddr=0x3ffb1b74 size=0x012bc (  4796) load
I (644) esp_image: segment 4: paddr=0x0005f838 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (659) esp_image: segment 5: paddr=0x0005fc40 vaddr=0x40080400 size=0x138b0 ( 80048) load
I (787) esp_image: segment 6: paddr=0x000734f8 vaddr=0x400c0000 size=0x00000 (     0) load
I (823) cpu_start: Pro cpu up.
I (824) cpu_start: Starting app cpu, entry point is 0x40080f04
0x40080f04: call_start_cpu1 at D:/msys32/home/Staubgeborener/esp/esp-idf/components/esp32/cpu_start.c:192

I (825) cpu_start: App cpu up.
I (833) heap_init: Initializing. RAM available for dynamic allocation:
I (854) heap_init: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (873) heap_init: At 3FFB7280 len 00028D80 (163 KiB): DRAM
I (892) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (911) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (931) heap_init: At 40093CB0 len 0000C350 (48 KiB): IRAM
I (950) cpu_start: Pro cpu start user code
I (1009) cpu_start: Starting scheduler on PRO CPU.
I (1013) cpu_start: Starting scheduler on APP CPU.
I (1013) example: Initializing SD card
I (1043) wifi: wifi firmware version: 0e33bec
I (1043) wifi: config NVS flash: enabled
I (1043) wifi: config nano formating: disabled
I (1043) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1053) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1083) wifi: Init dynamic tx buffer num: 32
I (1083) wifi: Init data frame dynamic rx buffer num: 32
I (1083) wifi: Init management frame dynamic rx buffer num: 32
I (1083) wifi: wifi driver task: 3ffbb23c, prio:23, stack:4096
I (1093) wifi: Init static rx buffer num: 10
I (1093) wifi: Init dynamic rx buffer num: 32
I (1103) wifi: Init rx ampdu len mblock:7
I (1103) wifi: Init lldesc rx ampdu entry mblock:4
I (1103) wifi: wifi power manager task: 0x3ffc05a0 prio: 21 stack: 2560
I (1113) wifi: wifi timer task: 3ffc160c, prio:22, stack:3584
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:5108
load:0x40078000,len:0
load:0x40078000,len:12652
entry 0x40078f44
```

I also tried a new usb cable and another usb host. But the problem seems to be wifi and sdmmc usage?

negativekelvin, 2017-08-01:
Have you tried to wait to init SD until after WiFi init is finished ?

Staubgeborener, 2017-08-02:
It's weird. I tried the following: 
SD init, 1 second delay, wifi init -> brownout
wifi init, 1 second delay, SD init -> brownout
just pressing boot button on dev board -> SD init failed. 
Disconnected IO0 and IO2 and wired those two IOs again -> it's working 

I will left this thread open, maybe this issue will happen again in the next days / hours. When nothing happens for one week, i will close this thread.

Staubgeborener, 2017-08-02:
I don't know if this belongs in this thread, but i think so, 'cause this issue has often somethings to do with power supply. 

I deactivate browning in ``make menuconfig``. After that everything works fine for hours. But now, from one time to another without changing my circuit or my code, i'll get

``Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.``

Full error:
```c++
I (47) boot: ESP-IDF v3.0-dev-205-g9314bf0d-dirty 2nd stage bootloader
I (47) boot: compile time 19:47:23
I (59) boot: Enabling RNG early entropy source...
I (109) boot: SPI Speed      : 40MHz
I (148) boot: SPI Mode       : DIO
I (186) boot: SPI Flash Size : 4MB
I (224) boot: Partition Table:
I (258) boot: ## Label            Usage          Type ST Offset   Length
I (326) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (396) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (465) boot:  2 app0             OTA app          00 10 00010000 00140000
I (535) boot:  3 app1             OTA app          00 11 00150000 00140000
I (605) boot:  4 eeprom           Unknown data     01 99 00290000 00001000
I (674) boot:  5 spiffs           Unknown data     01 82 00291000 00169000
I (744) boot: End of partition table
I (784) boot: Disabling RNG early entropy source...
I (835) boot: Loading app partition at offset 00010000
I (888) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x11890 ( 71824) map
I (1101) esp_image: segment 1: paddr=0x000218b8 vaddr=0x3ffc0000 size=0x02eec ( 12012) load
I (1125) esp_image: segment 2: paddr=0x000247ac vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (1166) esp_image: segment 3: paddr=0x00024bb4 vaddr=0x40080400 size=0x0b45c ( 46172) load
I (1340) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x4e844 (321604) map
0x400d0018: _flash_cache_start at ??:?

I (1923) esp_image: segment 5: paddr=0x0007e864 vaddr=0x4008b85c size=0x08f08 ( 36616) load
0x4008b85c: wr_bt_tx_atten at ??:?

I (1998) esp_image: segment 6: paddr=0x00087774 vaddr=0x400c0000 size=0x00000 (     0) load
I (2036) cpu_start: Pro cpu up.
I (2036) cpu_start: Starting app cpu, entry point is 0x40080f58
0x40080f58: call_start_cpu1 at D:/msys32/home/Staubgeborener/esp/esp-idf/components/esp32/cpu_start.c:192

I (2082) cpu_start: App cpu up.
I (2121) heap_init: Initializing. RAM available for dynamic allocation:
I (2185) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (2241) heap_init: At 3FFC8180 len 00017E80 (95 KiB): DRAM
I (2299) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (2358) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (2419) heap_init: At 40094764 len 0000B89C (46 KiB): IRAM
I (2477) cpu_start: Pro cpu start user code
I (2636) cpu_start: Starting scheduler on PRO CPU.
I (2641) cpu_start: Starting scheduler on APP CPU.
E (13241) sdmmc_cmd: sdmmc_card_init: send_scr returned 0x109
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400014fd  PS      : 0x00060b30  A0      : 0x80106688  A1      : 0x3ffd01a0
A2      : 0x00000030  A3      : 0x0000002c  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd0160
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x00000004  A13     : 0x00000001
A14     : 0x3ffd0120  A15     : 0x00000004  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000030  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400014fd:0x3ffd01a0 0x40106685:0x3ffd01b0 0x40115aa9:0x3ffd04c0 0x401165ab:0x3ffd0510 0x4011569a:0x3ffd0530 0x4011583f:0x3ffd05d0 0x40112f00:0x3ffd05f0
0x40106685: _vfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529

0x40115aa9: fprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fprintf.c:56

0x401165ab: sdmmc_card_print_info at D:/msys32/home/Staubgeborener/esp/esp-idf/components/sdmmc/sdmmc_cmd.c:244

0x4011569a: _Z15Initialisierungv at D:/msys32/home/Staubgeborener/esp/esp-idf/promiscuous_mode_esp32/main/promiscuous_mode_esp32.cpp:226

0x4011583f: _Z4loopv at D:/msys32/home/Staubgeborener/esp/esp-idf/promiscuous_mode_esp32/main/promiscuous_mode_esp32.cpp:291

0x40112f00: _Z8loopTaskPv at D:/msys32/home/Staubgeborener/esp/esp-idf/promiscuous_mode_esp32/components/arduino/cores/esp32/main.cpp:18 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:5108
load:0x40078000,len:0
load:0x40078000,len:12664
entry 0x40078f50
```

I don't have IO0 and IO2 connected, so i'll get
``E (3801) sdmmc_cmd: sdmmc_card_init: send_scr returned 0x109``
when connecting those sdmmc works, but error message is the same.

And yes, i use core 1 with ``xTaskCreatePinnedToCore()`` to work with time intensive operations. And this worked all the time.

@negativekelvin

igrr, 2017-08-02:
Sorry, I don't think the brownout on a particular board when you connect an SD card to it belongs to this issue tracker. 

Brownout detector indicates that the power supply on the board is not okay for the task; disabling it merely masks the issue, you may get crashes and other odd behavior.

closed 2017-08-02 by igrr

Staubgeborener, 2017-08-02:
I found out, that i'll get the brownout message when using sdmmc and wifi. As soon as wifi is initialized with ``esp_wifi_start()`` the brownout message appears. 
Only wifi -> worked. Only sdmmc -> worked. Wifi + sdmmc -> brownout.