espressif/esp-idf #869: i2c errors when using adc2

jmorsap, 2017-08-02:
I'm new to esp32/esp-idf and am moving a large embedded project from an atmega328 in the arduino environment to an esp32 wroom32 using esp-idf.

I'm aware that adc2 is not supported and plan on eliminating it in the next design iteration, but I needed the functionality for our current design and so added the ADC2 driver according to ESP32DE's procedure found here https://github.com/espressif/esp-idf/issues/461 

The adc2 works fine in its own program, as does I2C. 

When I configure the two settings below, the I2C ceases to work and will not work again until a hardware reset.

```
    adc2_config_width(ADC_WIDTH_12Bit);
    adc2_config_channel_atten(ADC2_TEST_CHANNEL,ADC_ATTEN_11db);
```

Is there any way to undo these two lines of code or reset these registers after I finish using adc2 so that I can use the I2C again?

CarlosGS, 2017-08-02:
Hi! I've used I2C + ADC2 without a problem (apart from
https://github.com/espressif/arduino-esp32/issues/349 ).
What pins are you using for I2C?

jmorsap, 2017-08-02:
Hello CarlosGS, Thanks for the reply

I2C does not work with the following config
`#define I2C_EXTERNAL_MASTER_SCL_IO    15    /*!< gpio number for I2C master clock */

#define I2C_EXTERNAL_MASTER_SDA_IO    4    /*!< gpio number for I2C master data  */

#define I2C_EXTERNAL_MASTER_NUM I2C_NUM_0   /*!< I2C port number for master dev */

#define I2C_EXTERNAL_MASTER_TX_BUF_DISABLE   0   /*!< I2C master do not need buffer */

#define I2C_EXTERNAL_MASTER_RX_BUF_DISABLE   0   /*!< I2C master do not need buffer */

#define I2C_EXTERNAL_MASTER_FREQ_HZ    100000     /*!< I2C master clock frequency */`

Though it does appear to work on GPIO 27 and 14.

I have also found that

`    adc2_config_channel_atten(ADC2_TEST_CHANNEL,ADC_ATTEN_11db);`

is the problematic function, but have not been successful at pinpointing the exact piece of code in this function that is preventing me from using I2C on pins 15 and 14. I suspect that it is something in the rtc_module.c file.

negativekelvin, 2017-08-04:
ADC2_TEST_CHANNEL = ADC2_CHANNEL_9 ?

jmorsap, 2017-08-04:
yes, 

`#define ADC2_TEST_CHANNEL (9)`

negativekelvin, 2017-08-04:
I think you should delete these 3 lines https://github.com/espressif/esp-idf/pull/466/files#diff-9934b2bd0fdda201ccf125abd90da487R662

FayeY, 2017-08-09:
Hi @jmorsap , did the above suggestion help you resolve the issue?

jmorsap, 2017-09-12:
Hey I apologize for not responding. We cut a trace on our boards and wired the signal to an ADC1 IO. Our new boards don't use ADC2 for anything, so I haven't tried this out yet. I'll let you know if I get a chance. Has anyone else tried this yet?

closed 2017-10-18 by FayeY