espressif/esp-idf #870: Can't load the OTA partition!

FFtust, 2017-08-02:
I used the OTA example to make a ota update, it seemed that the file had been writted to the flash, but 
later the printed information showed "cannot load", the details is below:

     [0;32mI (34703) ota: Have written image length 1339314[0m
     [0;32mI (34713) ota: Have written image length 1340338[0m
     [0;32mI (34723) ota: Have written image length 1341362[0m
     [0;32mI (34733) ota: Have written image length 1342074[0m
     [0;32mI (34743) ota: Have written image length 1343098[0m
     [0;32mI (34743) ota: Have written image length 1344122[0m
     [0;32mI (34743) ota: Have written image length 1344346[0m
     [0;32mI (34753) ota: Have written image length 1345370[0m
     [0;32mI (34753) ota: Have written image length 1346394[0m
     [0;32mI (34753) ota: Have written image length 1347106[0m
     [0;32mI (34763) ota: Have written image length 1348130[0m
     [0;32mI (34763) ota: Have written image length 1349154[0m
     [0;32mI (34763) ota: Have written image length 1349328[0m
     [0;32mI (34763) ota: Connection closed, all packets received[0m
     [0;32mI (34763) ota: Total Write binary data length : 1349328[0m
     [0;32mI (34793) ota: Prepare to restart system![0m
     I (34793) wifi: state: run -> init (0)
     I (34793) wifi: pm stop, total sleep time: 0/28207109

     I (34793) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
     E (34803) wifi: esp_wifi_connect 816 wifi not start
     I (35793) wifi: lmac stop hw txq
     [0;32mI (13) boot: ESP-IDF v1.0-alpha1-3-g21fc3c9-dirty 2nd stage bootloader[0m
     [0;32mI (14) boot: compile time 12:10:28[0m
     [0;32mI (14) boot: Enabling RNG early entropy source...[0m
     [0;32mI (30) boot: SPI Speed      : 40MHz[0m
     [0;32mI (43) boot: SPI Mode       : DIO[0m
     [0;32mI (55) boot: SPI Flash Size : 4MB[0m
     [0;32mI (67) boot: Partition Table:[0m
     [0;32mI (79) boot: ## Label            Usage          Type ST Offset   Length[0m
     [0;32mI (101) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
     [0;32mI (124) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
     [0;32mI (147) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
     [0;32mI (171) boot:  3 factory          factory app      00 00 00010000 00080000[0m
     [0;32mI (194) boot:  4 ota_0            OTA app          00 10 00090000 001c0000[0m   
     [0;32mI (217) boot:  5 ota_1            OTA app          00 11 00250000 00010000[0m
     [0;32mI (241) boot: End of partition table[0m
     [0;32mI (254) boot: Disabling RNG early entropy source...[0m
     [0;32mI (271) boot: Loading app partition at offset 00090000[0m
     [0;32mI (307) boot: segment 0: paddr=0x00090018 vaddr=0x3fff0008 size=0x00008 (     8) load[0m
     [0;31mE (315) boot: Segment 0 end address 3fff0010 overlaps bootloader stack 3ffe3cd0 - can't       load[0m
user code done
 
By the way, I changed the partition table because trhe file I want to update is bigger than 1M. And after I changed the partition, the hello_world.bin that  the ota example used worked well, but the file I tried to update  failed.

FHFS, 2017-08-04:
Why does it say:
>  [0;32mI (13) boot: ESP-IDF v1.0-alpha1-3-g21fc3c9-dirty 2nd stage bootloader[0m

Because my ESP32 says: 
> I (46) boot: ESP-IDF v3.0-dev-188-g851e4921-dirty 2nd stage bootloader

FayeY, 2017-08-09:
Hi @FFtust , can you have a try with the latest ESP-IDF? Is this issue still unsolved?

FFtust, 2017-08-29:
I am so sorry that I  have forgotten to check this page. 
A wrong ".bin" file casue this problem,  I used the hex file that generated by  micropython-esp32 project, it contains some other information at the beginning of this file, and the bootloader  of esp_idef didn't recognize it.

closed 2017-11-21 by FayeY