espressif/esp-idf #872: Use of xTaskNotify cause reboots

nilp0inter, 2017-08-03:
Hi,

I am trying to use xTaskNotify in my application but from time to time I receive this messages and the board reboots:

```
/esp/esp-idf/components/freertos/./tasks.c:4713 (xTaskNotify)- assert failed!
abort() was called at PC 0xI (75597) murmursocks: Queued RX packet40085de8 
on core 1

Backtrace: 0x40086b88:0x3ffc9e70 0x40086c87:0x3ffc9e90 0x40085de8:0x3ffc9eb0 0x400e00ee:0x3ffc9ed0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

```

Any clue of what's happening?

Thank you!

negativekelvin, 2017-08-03:
I think it is marked as untested smp function so it is expected to be experimental in dual core mode.

nilp0inter, 2017-08-03:
Yes, you are right @negativekelvin this function is marked as untested.

These "untested" functions are planned to be tested and adapted to work on the esp32?

HubbyGitter, 2017-08-04:
@nilp0inter Using `vTaskNotifyGiveFromISR` and `ulTaskNotifyTake` without problems. The assertion is not because you have the config option "Halt when an SMP-untested function is called" enabled, right?

nilp0inter, 2017-08-04:
@HubbyGitter No, "Halt when an SMP-untested function is called" is disabled.

I am using xTaskNotifyGive in one task and ulTaskNotifyTake in another, and getting this error randomly under heavy load.

I'll try vTaskNotifyGiveFromISR and be right back with the results.

nilp0inter, 2017-08-04:
Yeah, vTaskNotifyGiveFromISR gave me the same error:

```
/esp/esp-idf/components/freertos/./tasks.c:4900 (vTaskNotifyGiveFromISR...
- assert failed!
abort() was called at PC 0x40085de8 on core 1

Backtrace: 0x40086b60:0x3ffc9e90 0x40086c5f:0x3ffc9eb0 0x40085de8:0x3ffc9ed0 0x400e00d9:0x3ffc9ef0

Rebooting...

```

Dazza0, 2017-10-19:
@nilp0inter `xTaskNotify` will check if the task handle parameter provided is NULL by executing `configASSERT( xTaskToNotify );`

Just to make sure that this isn't the source of the abort, could you check to make sure the task handles that are passed into `xTaskNotify`  are not null. For example...
```
if(handle != NULL){
    xTaskNotify(handle, 0xFF, eSetValueWithOverwrite);
}else{
    printf("NULL pointer\n");
    fflush(stdout);
}
```

manoj153, 2019-05-10:
yes I am facing this issue. Any fix for this ?

Dazza0, 2019-05-10:
@manoj153 could you provide a backtrace of your crash showing which line of `xTaskNotify()` your program is crashing in. Also, could you also check you aren't passing a `NULL`  task handle to `xTaskNotify()`?

Dazza0, 2019-08-19:
@manoj153 Closing due to lack of updates. Please feel free to open a new issue if you are still facing this problem and suspect it is a bug in ESP-IDF.

closed 2019-08-19 by Dazza0