espressif/esp-idf #876: lmacProcessTxRtsError 1600

nkolban, 2017-08-04:
I'm building and testing a large application on the new WROVER module using the ESP-IDF and Xtensa toolchain recommended for use with WROVER as of this date.  My application seems to run just fine but seems to nondeterministically crash with the following messages:

```
W (50503) wifi: tx rts error 0x17

E (50503) wifi: lmac.c lmacProcessTxRtsError 1600

Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
```

I feel that I am constantly invoking ESP-IDF and FreeRTOS functions so am not sure how to interpret this story.

liuzfesp, 2017-10-10:
@nkolban could you let me know how to reproduce this issue? It will be better if you can show me the sdkconfig?

nkolban, 2017-10-10:
I will try and make some time to recreate ... thanks for the pointer on the sdkconfig ... I'll check and see if the problem still persists.  70 days is a long time ago and all context on the issue is gone from my mind ... I'll need to get into the habit of archiving the code for issues ready for a historic recreate.

FayeY, 2017-11-21:
Please feel free to reopen it if it still can be reproduced.

closed 2017-11-21 by FayeY