espressif/esp-idf #877: can't connect wifi when wifi sniffer open

zhao007z8, 2017-08-04:
i use this code to sniffer mac addr. 
https://github.com/lpodkalicki/blog/tree/master/esp32/016_wifi_sniffer
but when want to connect wifi. it tell me "SYSTEM_EVENT_STA_DISCONNECTED"

code url：
https://github.com/lpodkalicki/blog/tree/master/esp32/016_wifi_sniffer

zhao007z8, 2017-08-04:
my code:https://pan.baidu.com/s/1i5OOpxj

negativekelvin, 2017-08-04:
If you switch channel you will disconnect from AP

zhao007z8, 2017-08-04:
/*while (true) {
		gpio_set_level(LED_GPIO_PIN, level ^= 1);
		vTaskDelay(WIFI_CHANNEL_SWITCH_INTERVAL / portTICK_PERIOD_MS);
		wifi_sniffer_set_channel(channel);
		channel = (channel % WIFI_CHANNEL_MAX) + 1;
    	}*/

when i remove esp_wifi_set_channel  still have the issue

liuzfesp, 2017-08-15:
Hi @zhao007z8 you configure the WiFi Mode to NULL mode via ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_NULL) ); in wifi_sniffer_init, it should be WIFI_MODE_STA.

FayeY, 2017-09-05:
Hi @zhao007z8 , did the above suggestion help you resolve the issue?

jakkra, 2017-09-22:
Had the same problem, the problem is that you continuously change the sniffer channel. This causes the AP to disconnect. I managed to work around it and to also allow me to scan/connect/keep tcp connections running, and at the same time using the sniffer. You will however miss some packets, but I don't think it's possible on the ESP32 not to, if you want to be connected to an AP and sniff different channels at the same time. I did it like this:

0. Set sniffer channel
1. Enable sniffer
2. Wait 400ms (time_sniffing)
3. Disable sniffer
4. Call esp_wifi_set_channel(channelConnectedToApOn)
5. Wait 1000ms (time_not_sniffing)
6. Go to 0.

You can fine tune the time_sniffing and the time_not_sniffing to maximise the time you can sniff on a channel which isn't the channel your connected AP is on. Sniffing to long on a channel which isn't the one connected to your AP, will result in a disconnect.

closed 2018-01-11 by FayeY